arendst/Tasmota #372: PowerOnState=1 and stat message

a-sp, 2017-04-22:
It's my understanding that when the switch is powered on with a power cycle (with PowerOnState set to 1), the stat message is not generated.
Would it be possible to send the stat message also when the switch is automatically set to ON after a power outage?

dy008, 2017-04-23:
Hi，I have same qusion!
but i just change sonoff.ino it:
#define APP_POWER              1            // Default saved power state On
and power on it sent PowerOnState set to 1!
just do it!

arendst, 2017-04-24:
The stat message is generated ten seconds after power on allowing all communication and sensors to settle.

davidelang, 2017-04-26:
it sounds like this is not needed, the message is already generated. reopen this issue if you disagree.

closed 2017-04-26 by davidelang

a-sp, 2017-05-06:
Would it be possible to send the stat message immediately? After few seconds, my automation platform (home assistant) override the PowerOnState=1 and powers off the sonoff.

davidelang, 2017-05-07:
It waits a few seconds before sending the first message to give everything time 
to settle.

This sounds like you need to change home assistant to not change it to what it 
considers the prior state when it comes up.

On Sat, 6 May 2017, a-sp wrote:

> Would it be possible to send the stat message immediately? After few seconds, my automation platform (home assistant) override the PowerOnState=1 and powers off the sonoff.