arendst/Tasmota #373: Sonoff RF problems in wireless

adamast0r, 2017-04-23:
I am having problems connecting to the wireless with my Sonoff RF. I have created 2 wireless networks (indebuurt1,indebuurt2), the serial console has:

00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_E9C7BF, GroupTopic sonoffs) Version 4.1.3
00:00:00 Wifi: Connecting to AP1 indebuurt1 in mode 11N as sonoff-1983...
00:00:15 Wifi: Connect failed with AP timeout
00:00:15 Wifi: Connecting to AP2 indebuurt2 in mode 11N as sonoff-1983...
00:00:30 Wifi: Connect failed with AP timeout
00:00:30 WPSconfig: Failed to start

I also cannot start with 4 short presses the Access Point mode or at least I cannot get the serial log of that and I get no wireless AP, any idea what can be wrong ? 

Not sure if this is a hardware problem or a specific bug related with the Sonoff RF, since I can use non RF modules without any of these issues...

davidelang, 2017-04-23:
try increasing your logging debug level and make sure that it's using the 
correct password for the network

johntdyer, 2017-12-15:
I think I am seeing the same issue ( #1366  )