arendst/Tasmota #374: Wireless Password Leak

adamast0r, 2017-04-23:
I have been using my sonoff basic version without any issues for the last weeks. I noticed something strange some days ago. I could not see the sonoff connected to my network, it seems that it went to the AP mode without any intervention from myself (bug?). This is a problem since it was in AP mode an anyone could:

1) connect to the sonoff (192.168.4.1) 
2) view the source code of the page and get my network wireless password
3) connect with that password to my wireless network

As a security measure I don't think the previous passwords should be available when the source code of the page is seen.

davidelang, 2017-04-23:
you can disable the multipress functions.

the password is also visible on the web console and via mqtt.

The mindset is that if you have people in your house hitting the buttons or 
otherwise sending the devices commands to switch to AP mode, you have problems 
far more severe than the leakage of the password.

adamast0r, 2017-04-23:
The problem is that nobody pressed the buttons. 

Because you have passwords in other places does not make the issue less serious, maybe they should also be fixed.

davidelang, 2017-04-23:
It's very useful for troubleshooting, a config option to mask them is an option 
(but takes code and memory that are both in short supply)

If you have a device switching to AP mode without you telling it to, then you 
either have it configured to do so when it looses it's wifi connection, or 
someone is telling it to do so without you knowing about it. you should 
investigate your logs (and your config) to know which is happening.

fkahhaleh, 2017-04-23:
Not to hijack the thread, but David is it possible to not have the sonoff go into AP unless a physical hardware button is pressed? (and nothing else?)

Perhas Adam can start with that route and observe the behavior.
I am waiting on my units to arrive and looking forward to working on and using this firmware :)

davidelang, 2017-04-23:
yes, change the WIFI_CONFIG_TOOL settings to something other than WIFI_MANAGER

adamast0r, 2017-04-26:
@davidelang 
"It's very useful for troubleshooting, a config option to mask them is an option
(but takes code and memory that are both in short supply)"

To be honest I think having a option is overkill, just don't write the current password in the web interface, no extra memory is required.  If the person want to change the password it will still be possible, there is no reason to have the current password in the webinterface.

In the current design the user can't even verify in a convenient way that he put the correct password, since the password is masked by the browser.

richrd, 2017-11-01:
This does seem to be an issue. If I set the mode to `WIFI_RESTART` in the `user_config.h` file it works as expected. Then if I press the button 4 times it starts wifi manager as expected. Everything ok at this point.
After that though, it always starts wifi manager instead of wifi restart every time it loses wifi. I only want it to start wifi manager when I press the button. Not automatically. Is there a workaround?

lal12, 2018-02-10:
I noticed the same thing. When e.g. my Router updates its ap is down, then the Sonoff opens its own WiFi-AP, which is a very nice feature. But everyone could now read my wifi password.
It is not that uncommon for a router to shut down once in a while for an update. So I guess it really should be handled as @adamast0r proposed.

ascillato, 2018-02-10:
@lal12
This issue is old and it is fixed now #1527
This issue should be closed

lal12, 2018-02-10:
@ascillato Ok, thanks but I guess you referenced the wrong Issue: #1587

ascillato, 2018-02-10:
Yes. You are right! That one I wanted to reference. Thanks!

closed 2018-06-02 by ascillato