arendst/Tasmota #375: Unclear Syntax in console windows setting up switchmode

the-Leitz, 2017-04-23:
i'm trying to change te SwitchMode in the console window. but nothing seems to work.
i have tried
Switchmode=3
Swtichmode3
Switchmode 3
Cmd/Set/Switchmode3 etc
nowhere is a good description of how the syntax is used
are they the same as the mqtt commands do i have to use slashes , etc
i just want to set it up as a pushbutton, editing the firmware seems not the way to go, but there are settings there (why ?), from all the information i can find, i think the only way to configure this is inside the console.
i'm not using any server to control it.
Can somone tell explain how to correctly use the syntax in the console ?

davidelang, 2017-04-23:
have you gone through the wiki page?

https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

the-Leitz, 2017-04-23:
Yep if i give the command switchmode=3 i get an error (unknown command) i tried it with cmnd/set etc but i always get the same error

davidelang, 2017-04-23:
try "switchmode1 3"

the-Leitz, 2017-04-23:
Nope same error
13:54:24 CMND: Switchmode1 3
13:54:24 RSLT: DataCb Topic sonoff, Group 0, Index 1, Type SWITCHMODE, Data
3 (3)
13:54:24 RSLT: RESULT = {"Command":"Unknown"}
13:55:27 CMND: Switchmode 1 3
13:55:27 RSLT: DataCb Topic sonoff, Group 0, Index 1, Type SWITCHMODE, Data
1 3 (1 3)
13:55:27 RSLT: RESULT = {"Command":"Unknown"}

dy008, 2017-04-23:
I think the switchmode and 1 no need Space!

the-Leitz, 2017-04-23:
I tried that same result :(

davidelang, 2017-04-23:
are you sure you have a gpio configured as a switch? until you do that, the 
switchmode commands will not work.

the-Leitz, 2017-04-23:
Gpio ? It is just working fine as a toggle switch, but it needs to act like
a pushbutton

davidelang, 2017-04-23:
go to the module config page and look at the gpio settings there. you need to 
set the gpio that you have the switch connected to to be a switch, otherwise 
things aren't going to work.

the-Leitz, 2017-04-23:
I want the relay to act as a pushbutton, are we talking about the same
thing ?

davidelang, 2017-04-23:
>     <reply+0021b24ff80f804c88dd676b1a4014e30a5db665b89ed70892cf00000001151486c
>     f92a169ce0d5437f5@reply.github.com>
>     setting up switchmode (#375)
>
> I want the relay to act as a pushbutton, are we talking about the same
> thing ?

in sonoff terms, a led or relay is an output, a switch or button is an input. if 
you want to have something tell the sonoff to do something, you need to 
configure it as a switch or button

if you want to configure the sonoff tell something else to change, you need to 
configure it as a LED or relay

so if you are hooking a switch up to the sonoff to tell it to turn the relay on 
and off, you need to confure the gpio that you wire the switch to as a switch or 
button. If you want it to change relay1 (the built-in relay), this will need to 
be switch1 (since button1 is already defined)

does this clarify things?

the-Leitz, 2017-04-23:
Not really, are you saying that the switchmode command does not change the behavior of the relay ? But why gives it an error (unknown command)

davidelang, 2017-04-23:
the switchmode command changes what happens when you press a switch.

to set a switchmode, you have to first have a switch defined to set the mode on.

so if you are wiring a switch to gpio14, then you first have to define gpio14 as 
switch1, then you can issue the switchmode1 command

the-Leitz, 2017-04-23:
:( is there any command to change the toggle in to a pushbutton ? (The
relay)

davidelang, 2017-04-23:
>     <reply+0021b24f07338841c44c48a6a7583239d064e3c94ed5d44e92cf00000001151499e
>     192a169ce0d5437f5@reply.github.com>
>     setting up switchmode (#375)
>
> :( is there any command to change the toggle in to a pushbutton ? (The
> relay)

you don't change the relay, you change the switch settings. This is what 
switchmode (and buttonmode for buttons) do.

but you have to have a switch defined before you can set it's mode.

have you defined a switch on the gpio yet?

the-Leitz, 2017-04-23:
Shouldn't i configure a button ? And which gpio should i choose ?

davidelang, 2017-04-23:
well, if you want to have it control relay1 (the built-in relay), you would need 
to configure button1, but since button1 is already defined (the button on the 
module) and you can't define two gpios to have the same function, you can't 
define button1. So you need to define switch1

As to which gpio you configure it on, that depends on which gpio you have wired 
it to. you need to configure the switch in software to match where you have 
attched the switch on the hardware.

the-Leitz, 2017-04-23:
No i do not want control the relay with an external switch, i want to user the webinterface . When i Press the toggle button, i want the relay to turn on and directly off again. So It works like a pushbutton . I understand now that the button options in the firmware are all about, external buttons you can connect to the module. So thats not the way to go. Do you have any idea How i can achieve this ?

davidelang, 2017-04-23:
so you aren't actually wiring up a physical switch?

you are just talking about changing the web interface?

There is no option to change the web interface, you would have to edit the 
webserver.ino source to add this functionality.
k

the-Leitz, 2017-04-23:
Yes i think i Will have to do some scripting, I don't know if this is possible

davidelang, 2017-04-23:
This is not scripting, it is modifying the C code in webserver.ino to add 
additional buttons to the html, and then to respond when those buttons are 
pressed.

the-Leitz, 2017-04-23:
No no, no extra buttons, just the behaviour, like ' if on then off ' I don't think it is possible in tasmota, thx for the help

davidelang, 2017-04-23:
that's what toggle is.

and no, tasmota does not give you any ability to redefine what the web interface 
does.

On Sun, 23 Apr 2017, the-Leitz wrote:

> No no, no extra buttons, just the behaviour, like ' if on then off ' I don't think it is possible in tasmota, thx for the help
>
>

the-Leitz, 2017-04-24:
i just want to say that i resolved it, i used the PulseTime Command and a single puls and it works :D

closed 2017-04-24 by the-Leitz

djdom111, 2017-07-21:
The Leitz, can you tell me how you used the pulse time command on the console. I need to do the same with sonoff as you wanted and did it. But for me doesn't work.

the-Leitz, 2017-07-24:
i'm sorry i can not remember the commands, but i used the pulstime command to give 1 pulse of 1 second that worked for me

tomasroud, 2017-09-15:
cmnd/YOURDEVICE/pulsetime 10
will do when you send cmnd/YOURDEVICE/POWER ON ... it stays on 1 second and then OFF relay and set state to POWER OFF