arendst/Tasmota #376: #define STA_SSID2 not recognised

alan-rpi, 2017-04-23:
I raised a thread here https://groups.google.com/forum/#!msg/sonoffusers/ytxx_v0_myw/xFbWzRG-BAAJ explaining that my wi-fi access point could not see the Sonoff S20. Please see that for details. 

My original settings were 
#define STA_SSID1              "abc"      // [Ssid1] Wifi SSID                   I've replace real SSID with abc       
#define STA_PASS1              "zzzzzzzz"   // [Password1] Wifi password
#define STA_SSID2              ""      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""      // [Password2] Optional alternate AP Wifi password
Even with the security disabled on abc the S20 was not making access.

I continue to investigate and opened a Putty terminal to the device. The log showed

`00:00:33 APP: Restarting
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
   ▒
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_9F4B86, GroupTopic sonoffs) Version 4.1.3
00:00:00 Wifi: Connecting to AP1   abc in mode 11N as sonoff-2950...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP2 indebuurt2 in mode 11N as sonoff-2950...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 WPSconfig: Active for 1 minute`
-----------------------------

Notice that AP2 is indebuurt2, the default setting before I set it to blank.

I then found a Edimax wifi dongle and set it up as an AP. I update tasmota to

-----------------------------
#define STA_SSID1              "abc"      // [Ssid1] Wifi SSID   <<<<< my TP-Link Ap
#define STA_PASS1              "zzzzzzzzzzz"   // [Password1] Wifi password
#define STA_SSID2              "xyz"      // [Ssid2] Optional alternate AP Wifi SSID   <<<<< my Edimax dongle AP
#define STA_PASS2              ""      // [Password2] Optional alternate AP Wifi password   << no pswd
-----------------------------

I reflashed the S20 and reran Putty. Now the log is

-----------------------------
00:00:33 APP: Restarting
ets Jan  8 2013,rst cause:1, boot mode:(3,7)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
   ▒
00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_9F4B86, GroupTopic sonoffs) Version 4.1.3
00:00:00 Wifi: Connecting to AP2 indebuurt2 in mode 11N as sonoff-2950...
00:00:07 Wifi: Connect failed as AP cannot be reached
00:00:07 Wifi: Connecting to AP1   abc in mode 11N as sonoff-2950...
00:00:14 Wifi: Connect failed as AP cannot be reached
00:00:14 WPSconfig: Active for 1 minute
-----------------------------

Still indebuurt2 is being used.

Thanks 
Alan

davidelang, 2017-04-23:
look at the troubleshooting page and the section on configs.

k

alan-rpi, 2017-04-23:
David, thanks for a quick response. I found the Troubleshooting page and don't feel any of them are prime targets for this issue (i.e. trying to use indebuurt2 as the SSID - where is the flashed code getting indebuurt2 from if it was changed in the source code to something else?) 

The  CFG_HOLDER  might be worth a go but what do I change it to? 

#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters

I couldn't find a supplier of the S20 in the UK so sourced from China and that took 2 weeks to arrive.

Alan

davidelang, 2017-04-23:
read the section on config problems. That explains in detail why changes to the 
code don't take effect as you expect.
k

davidelang, 2017-04-23:
On Sun, 23 Apr 2017, alan-rpi wrote:

> The  CFG_HOLDER  might be worth a go but what do I change it to?
>
> #define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters

sorry, I missed this part. you change it to anything else. all that matters is 
if it matches what was compiled in to the prior version that was flashed. If it 
matches, the config does not get wiped out, if not the existing config on the 
device is assumed to be corrupt/uninitialized and gets replaced by what's in 
user_config.h

alan-rpi, 2017-04-23:
Thanks David. Now working

Indeed CFG_HOLDER has to be updated (changed to any other value - I used today's date).

At first it seemed totally illogical that having changed the user config file and saved it then the changes were not uploaded. I guess the reason is that once it is up and running then further changes will be done via a browser using the web interface and you wouldn't want an out of date user config file destroying any changes made online. On the other hand, some changes must be via the user config file.

I've added a comment line before the CFG_HOLDER definition:

// **************** THE CFG_HOLDER VALUE MUST BE CHANGED (to any different value) FOR CHANGES TO BE UPLOADED ****************

Alan

closed 2017-04-23 by alan-rpi