arendst/Tasmota #380: PowerOnState seems to work accidentially. Proposal to fix is

stefanbode, 2017-04-24:
Hi all,

during my investigation to enable TASMOTA also for up to 32 relays, I found the PowerOnState functions and investigated them. The << and >> of the mask to get values only for the defined did not work with uint32_t. I did some debugging and found that (1<<Maxdevices)-1 did the job as well and reliable also with larger numbers. therefore see the proposed attached code changes. 

Also, the ESP.getResetReason should be configurable. Only if this value is there the toggle/on/off is working. e.g. the Wemos Mino never starts with this message. There it is "External system"

```
  if (sysCfg.module == MOTOR) sysCfg.poweronstate = 1;  // Needs always on else in limbo!
  if (ESP.getResetReason() == "Power On") {
  //if (sysCfg.poweronstate) {
    //snprintf_P(log, sizeof(log), PSTR("APP: maxdevices  %d"), Maxdevice);
    addLog(LOG_LEVEL_INFO, log);
    if (sysCfg.poweronstate == 0) {       // All off
      //snprintf_P(log, sizeof(log), PSTR("APP: Setting powermatrix all OFF 0x%8x"),power);
      power = 0;
      setRelay(power);
    }
    else if (sysCfg.poweronstate == 1) {  // All on
      power = (1<<Maxdevice)-1;
      //snprintf_P(log, sizeof(log), PSTR("APP: Setting powermatrix all ON 0x%8x"),power);
      setRelay(power);
    }
    else if (sysCfg.poweronstate == 2) {  // All saved state toggle
      power = (sysCfg.power & (1<<Maxdevice)-1 ) ^ 0xFFFFFFFF;
      if (sysCfg.savestate) setRelay(power);
      //snprintf_P(log, sizeof(log), PSTR("APP: Setting powermatrix all toggle 0x%8x"),power);
    }
    else if (sysCfg.poweronstate == 3) {  // All saved state
      power = sysCfg.power & (1<<Maxdevice)-1 ;
      if (sysCfg.savestate) setRelay(power);
      //snprintf_P(log, sizeof(log), PSTR("APP: Setting powermatrix all saved 0x%8x"),power);
    }
  } else {
    power = sysCfg.power & (1<<Maxdevice)-1 ;
    if (sysCfg.savestate) setRelay(power);
    //snprintf_P(log, sizeof(log), PSTR("APP: Setting powermatrix savestate 0x%8x, from 0x%8x, shifter is 0x%8x, maxdevice %d"),power, sysCfg.power, (1<<Maxdevice)-1 , Maxdevice);

  }
  addLog(LOG_LEVEL_INFO, log);
  blink_powersave = power;
```

CommodoreWhite, 2017-04-24:
Hi. I am encouraged by the results into investigating the power-on status returned by your Wemos Mino. You appear to be experiencing the same issue I have has with my v1.5 Electrodragons in that they also return "External System", see issue #284. I have not heard of other dragon owners experiencing this problem, perhaps this thread will help bottom out the issue.

Peter

raisude, 2017-04-24:
I have a sonoff sv. The relay works as a potential free contact. In my case it opens a sliding gate.I have also the problem that PowerOnState 0 does not do the right thing. After powering on the sonoff the power is On. So the gate opens.

Would be very nice if this problem could be fixed.

arendst, 2017-04-28:
@stefanbode thanks for your research. I will update mine with your suggestion regarding ``(1<<Maxdevice)-1``.

I'm still investigating the power on issue. I hoped to use a RAM value which would be different after power on than at restart but currently I fail to find a functional solution; I can't get a value on the heap which keeps it's state after restart...

mrmin24, 2017-05-03:
This does not seem to be fixed in latest release? or is it just me for wemos?

stefanbode, 2017-05-03:
I checked the code and it looked like it is fixed by access some global variables. Should work but haven't tried so far

arendst, 2017-05-03:
Remember that it should only trigger at power on OR external system. By design it should not trigger if you perform a restart.

mrmin24, 2017-05-03:
{"PowerOnState":1}

{"Started":"External System"}

still relays not on. keep old state 

Side note. This is with the PCF8574 8 relay configuration from @stefanbode and the normal wemos with 4 relays

arendst, 2017-05-03:
I do not support this feature. Over to @stefanbode ...

mrmin24, 2017-05-03:
sorry, to be clear. 

I tested **both** the options. Normal wemos and PCF8574. Same result

arendst, 2017-05-03:
How do you test my option? How do you disconnect and re-connect power to the wemos?

mrmin24, 2017-05-03:
I plug it out and back in again. USB and then get the {"Started":"External System"} reply

arendst, 2017-05-03:
Right. I'll test mine...

arendst, 2017-05-03:
I see a press on the reset button works fine and performs poweronstate correctly. Plugging USB out and then in again doesn't work although the restart message is the same. Somewhere in between power settling and poweronstate test something goes wrong on the Wemos only...

mrmin24, 2017-05-03:
my reset button also does not work

arendst, 2017-05-04:
The problem with the Wemos D1 mini is caused by the Serial RTS circuit which, when the host initialises the USB serial interface, lowers GPIO00 which in turn will toggle the relay. 

This is what happens after reinserting the USB cable in your COM port:
1) Wemos boots up, checks resetinforeason and performs task poweronstate
2) Host serial initialises and within a second the Wemos RTS line is pulled low resulting in GPIO00 pulled low which is recognised as a button press and toggles the relay.

The obvious solution is to power the (production) wemos D1 mini from an external power supply or use a USB cable without data lines as does the Sonoff SC.

mrmin24, 2017-05-04:
I powered the wemos from a Cellphone charger and used my SC cable. Still same result. 

00:00:07 MQTT: tele/42/INFO3 = {"Started":"External System"}
00:00:08 MQTT: ctrlIn/42/RESULT = {"POWER1":"OFF"}
00:00:08 MQTT: ctrlIn/42/POWER1 = OFF
00:00:08 MQTT: ctrlIn/42/RESULT = {"POWER2":"ON"}
00:00:08 MQTT: ctrlIn/42/POWER2 = ON
00:00:08 MQTT: ctrlIn/42/RESULT = {"POWER3":"ON"}
00:00:08 MQTT: ctrlIn/42/POWER3 = ON
00:00:08 MQTT: ctrlIn/42/RESULT = {"POWER4":"OFF"}
00:00:08 MQTT: ctrlIn/42/POWER4 = OFF

Then I tested baseline **(Program version	5.0.3)**   and it seems to be working correctly now on Wemos mini.

Thanks a lot

closed 2017-05-10 by arendst