arendst/Tasmota #381: Connecting to WIFI but no IP Addressed assigned

Cruisecontrol2016, 2017-04-24:
Hello,

I new on here and just started trying to flash my sonoff last week. I did manage to flash the software and I believe my board is working properly. I can see from watching the serial monitor that the board is operating. This issue I am getting is the local IP addressed being assigned is 255.255.255.255.

00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_3BC225, GroupTopic sonoffs) Version 4.1.3
00:00:00 Wifi: Connecting to AP1 ****** in mode 11N as sonoff-0549...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on sonoff-0549.local with IP address 255.255.255.255

I have tried setting a static IP address in the user_config.f as well as leaving it 0.0.0.0 for DHCP and I still have the same outcome. 

I also was watching for the IP address from my router but my router doesn't display the sonoff as a connected device. Any feedback would be greatly appreciated.

davidelang, 2017-04-24:
try increasing the logging level, but it sounds like it really isn't getting an 
IP address from your AP.

Cruisecontrol2016, 2017-04-25:
How do I increase the logging level?

davidelang, 2017-04-25:
either through the web gui (configuration, logging) or via the commands as 
documented at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Cruisecontrol2016, 2017-04-25:
I can't get onto the web gui because I can't connect to the board through my wireless router. I would appreciate a little more specifics on how to use the commands. This is my first time ever doing this. I am attempting to sent commands in the Arduino IDE software using the serial monitor.

davidelang, 2017-04-25:
if you type

seriallog 4

on the serial port, it will execute that command

if you had network access you could type the same thing on the web console

if you had network access you could send the same command via mqtt (with the 
cmnd/name/ prefix to address it to the correct module)

Cruisecontrol2016, 2017-04-25:
Alright. In the serial monitor window, I put the commands in the line at the top and hit send but nothing happened. It just kept giving me the same logs. It just keeps doing the following over and over.

00:01:39 MQTT: Attempting connection...
00:01:40 mDNS: Query done with 0 mqtt services found
00:01:46 MQTT: Connect FAILED to 192.168.13:1883, rc -2. Retry in 10 seconds
00:01:57 MQTT: Attempting connection...

Am I putting the commands in wrong or could it be a different issue?

davidelang, 2017-04-25:
if you just type seriallog it will show you the current log level
k

Cruisecontrol2016, 2017-04-25:
Thank you very much. I got the commands working. Below are the new logs. The difference I see is the Patch issue 2186.  

00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_3BC225, GroupTopic sonoffs) Version 4.1.3
00:00:00 Config: Saved configuration (1364 bytes) to flash at F8 and count 125
00:00:00 Wifi: Checking connection...
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to AP1 ****** in mode 11N as sonoff-0549...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on sonoff-0549.local with IP address 255.255.255.255

arendst, 2017-04-25:
When running try changing the ip address with command ``ipaddress1 x.x.x.x`` where x.x.x.x is a valid ip address for your network. Do the same for your gateway (router) with command ``ipaddress2 y.y.y.y`` and your netmask ``ipaddress3 z.z.z.z``. Perform a restart with ``restart 1``.

This way it should receive a static address.

What did you change in user_config.h other than your ssid and password? I noticed an invalid MQTT server IP address of 192.168.13 or stop changing logging output.

Cruisecontrol2016, 2017-04-25:
I changed all the things you suggested. My router now sees the sonoff, but I am not able to open the web gui. I also tried pinging my sonoff from my computer and the request timed out. 

The other things I changed in the user_config.h file are I added two forward slashes (//) to #define USE_WS2812 , #define USE_IR_REMOTE , #define USE_I2C , and #define USE_DOMOTICZ. I did not define anything for the MQTT server as I was trying to use as much of the original code as possible for troubleshooting. 

00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_3BC225, GroupTopic sonoffs) Version 4.1.3
00:00:00 Config: Saved configuration (1364 bytes) to flash at F7 and count 134
00:00:00 Wifi: Checking connection...
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to AP1 ***** in mode 11N as sonoff-0549...
00:00:01 Wifi: Checking connection...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Checking connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Checking connection...
00:00:03 Wifi: Connected
00:00:03 mDNS: Initialized
00:00:03 HTTP: Webserver active on sonoff-0549.local with IP address 192.168.1.102
00:00:04 MQTT: Attempting connection...
00:00:05 mDNS: Query done with 0 mqtt services found
00:00:11 MQTT: Connect FAILED to 192.168.13:1883, rc -2. Retry in 10 seconds
00:00:22 MQTT: Attempting connection...
00:00:23 mDNS: Query done with 0 mqtt services found

davidelang, 2017-04-26:
does your access point show you getting connected to it? until that happens, IP addresses are meaningless.

Cruisecontrol2016, 2017-04-26:
Yes. My access point shows me connected.

davidelang, 2017-04-26:
If you leave the IP address set at 0.0.0.0 do you connect and get a DHCP IP 
address?

Cruisecontrol2016, 2017-04-26:
Aright. I switched it back to an ip address 0.0.0.0 and I did get a DHCP IP address and I was finally able to get onto the web server. I really appreciate the help. Any idea what caused this issue for me?

davidelang, 2017-04-26:
I suspect that you did not have the wifi settings loading correctly. you may have had old settings in the image and then compiled with new ones, but not have done anything to override the existing config.

closed 2017-04-26 by davidelang

tinoadams, 2017-07-07:
Hi

I've gotten myself into the same state as @Cruisecontrol2016, exactly the way @davidelang suggested by having old settings on the eeprom.

My user_config.h has WIFI_IP_ADDRESS set to "0.0.0.0". Shouldn't an erase/default/restart by holding the button for 4 seconds reset the stored config to that value? Even after and erase the unit's IP remained 255.255.255.255 and did not try and get a new lease from my DHCP server.

Setting an static ipaddress followed by setting the ip address to "0.0.0.0" using the "ipadress" command did the trick and the unit fetched a new IP from DHCP after a restart.

I looked at CFG_Erase and CFG_Default and can't see anything obvious wrong with it... any pointers? 
Then again, I don't understand the whole slots/flash_rotate idea. I guess I have some Googling todo.

Cheers
Tino

Jonny-O, 2017-09-08:
I ran into this same problem while flashing a batch of controllers. I was using the exact same setup across all six devices and five of them worked without a problem, so I knew the config and process were both correct. But the sixth device kept claiming its IP to be 255.255.255.255 and also displayed a bunch of "checkered-flag" characters during startup where parameter data should have been, as if something was corrupted.

I was able to fix this by changing the 'CFG_HOLDER' value in the user_config.h file from 0x20161209 to 0x20161208. Re-flashed the device and the problem disappeared. I then changed the variable back to its original value (didn't want to take any further chances at bricking the device) and re-flashed again, and it's now right as rain.

YMMV. I'm not a programmer and took the chance only because I was dealing with a device that's so inexpensive it's almost disposable. But it worked for me.

ssjha, 2017-09-20:
Jonny-O, thanks for your experiment. I was also facing the same issue and changing the CFG_HOLDER value did the trick for me. I was trying to flash a sonoff basic module.

orsgah, 2017-09-24:
Jonny-O, thanks s20 stop working with 255.255.255.255 web servers after of 2 weeks normal work, reflash with  CFG_HOLDER 0x20161208 help me.