arendst/Tasmota #382: Upgrade to 4.2.0

aferraz, 2017-04-24:
Hi,
 I upgrade from 4.1.3 to 4.2.0 a sonoff pow, 3 electrodragon and 1 nodemcu. Except the POW in all other cases I have lose my configuration and the electrodragon/esp startup with the user_config data. I try via firmware upload and ota, same result.

Now on the electrodragon I also have this situation:
21:18:04 CMND: power1
21:18:05 MQTT: /esp12/RESULT = {"POWER1":"POWER"}
21:18:05 MQTT: /esp12/POWER1 = POWER
21:18:08 CMND: power2
21:18:08 MQTT: /esp12/RESULT = {"POWER2":"POWER"}
21:18:08 MQTT: /esp12/POWER2 = POWER

How I can revert to ON/OFF state ?

Thanks

aferraz, 2017-04-24:
Sorry, found the solution in StateText1 and StateText2. But what about the configuration lost ?

arendst, 2017-04-24:
How did you upgrade? Did you use the new linker? If so then please keep using the old linker. I removed the new linker until further notice as the this version (4.2.0) has to be installed with the old linker first before the new can be used.

aferraz, 2017-04-24:
Hi @arendst, I have compiled the source as always. Non change IDE side.

aferraz, 2017-04-24:
I've tried now with the old linker with the same results. The configuration of the item is gone and was taken from user_config.
Tried on electrodragon

arendst, 2017-04-25:
What happens if you go back to version 4.1.3? Are the configuration items also gone or do they re-appear?

Version 4.2.should have copied the settings from spiffs flash to eeprom flash. It does this by comparing the CFG_HOLDER value in spiffs flash with the value in user_config.h. If they miscompare then the result you noticed will occur as it will load default settings in eeprom.

aferraz, 2017-04-25:
I'll try and give you a feedback

arendst, 2017-04-25:
No need to test. Problem is caused by residue data already present in the eeprom area. I narrowed the test window to limit the problem you have encountered. Sorry for the inconvenience. Next version will hopefully work better but that's of no use to you anymore...

aferraz, 2017-04-25:
No problems, thanks for your hard and very good work

aferraz, 2017-04-25:
Hi @arendst, 
I have installed 5.0 on an nodemcu but I cant see more free program space:
My values after flashing:
Flash size	4096kB
Program flash size	1024kB
Program size	439kB
Free program space	564kB
Free memory	29kB

I have compiled the firmware with PlatformIO
However, also in this case, I lost the settings, I recovered from the backup

arendst, 2017-04-26:
You see the gain only in Free program space which is incremented by 64k allowing me to free up Free memory with 1k by moving all static data to Program.

Sorry the settings were lost once again. All my 15 devices worked fine...

Did you compile your nodemcu code for 1G or 4G flash size? If it was 4G before than this is expected as the spiffs area is in another location between 1G and 4G images.

aferraz, 2017-04-26:
It's true, before was 4G.

closed 2017-04-26 by aferraz