arendst/Tasmota #383: PowerOnState

mrmin24, 2017-04-24:
On some devices like the Wemos the power on telemetry is {"Started":"External System"}. 

The powerOnState however checks for (ESP.getResetReason() == "Power on") before using the settings for the command. 

Thus the powerOnState does not work on all devices correctly. Works on SV for instance, but not Wemos

davidelang, 2017-04-26:
I'm curious as to why there is a need to do different things depending on the reason for the restart?

mrmin24, 2017-04-26:
Not sure, but maybe on soft reset the relays should not switch

arendst, 2017-04-28:
Will test for "External System" too when powered on in next version.

CommodoreWhite, 2017-05-01:
I have to say that my workaround was to change the code so that, no matter what the reason for the restart, the poweronstate setting was always honoured. This mean that in the event that a catastrophic power failure that took out my Mqtt services, the lights would always go on day or night.

CommodoreWhite, 2017-05-02:
Btw. TASMOTA v5.0.1 works a treat on my dragons. Thanks.

mrmin24, 2017-05-02:
Might be worth having this as a User Definable option?

mrmin24, 2017-05-03:
I will close this issue as it is duplicate for #380. I think we opened it very close together

I made a comment that I do not think it is working 100% yet in #380

thanks

closed 2017-05-03 by mrmin24