arendst/Tasmota #385: Updated to 4.1.3 on one of my Wemos modules and tele/cmnd/etc have been stripped

mihalski, 2017-04-25:
Here is an example of what I mean:
```
00:00:00 APP: Project sonoff Bedroom (Topic bedroom, Fallback DVES_1922AE, GroupTopic sonoffs) Version 4.1.3
00:00:00 Wifi: Connecting to AP1 DobNet in mode 11N as bedroom-0686...
00:00:05 Wifi: Connected
00:00:05 mDNS: Initialized
00:00:05 HTTP: Webserver active on bedroom-0686.local with IP address 192.168.1.84
00:00:06 MQTT: Attempting connection...
20:05:18 mDNS: Query done with 0 mqtt services found
20:05:18 MQTT: Connected
20:05:18 MQTT: /bedroom/LWT = Online (retained)
20:05:18 MQTT: Sonoff/bedroom/POWER = 
20:05:18 MQTT: /bedroom/INFO1 = {"Module":"WeMos D1 mini", "Version":"4.1.3", "FallbackTopic":"DVES_1922AE", "GroupTopic":"sonoffs"}
20:05:18 MQTT: /bedroom/INFO2 = {"WebserverMode":"Admin", "Hostname":"bedroom-0686", "IPaddress":"192.168.1.84"}
20:05:18 MQTT: /bedroom/INFO3 = {"Started":"Software/System restart"}
20:05:26 MQTT: /bedroom/STATE = {"Time":"2017-04-25T20:05:26", "Uptime":0, "Vcc":2.748, "Wifi":{"AP":1, "SSID":"DobNet", "RSSI":78, "APMac":"80:2A:A8:8A:21:B3"}}
20:05:26 MQTT: /bedroom/SENSOR = {"Time":"2017-04-25T20:05:26", "Switch1":"POWER", "BME280":{"Temperature":27.4, "Humidity":52.9, "Pressure":1008.1}, "BH1750":{"Illuminance":0}}
```
Has anyone encountered this before and/or know how to solve it?

arendst, 2017-04-25:
Nice to know from what version you came from but since version 4.1.0 these items have been changed to user configurable BUT allow only up to 10 chars max. See another issue somewhere around.

mihalski, 2017-04-25:
This particular device came from 4.0.5.

mihalski, 2017-04-26:
Resetting the configuration and reconfiguring from scratch solved the problem.

closed 2017-04-26 by mihalski