arendst/Tasmota #386: Changes to cause WS2812 strip switching to be independent from relay/power

joecotton, 2017-04-25:
The WS2812 module checks and modifies the global power variable, which means that sending the power on or off commands also changes the WS2812 strip. The patch below causes this to happen only if the module is SONOFF_LED.

Ref: #85 

```
--- a/sonoff/xdrv_ws2812.ino
+++ b/sonoff/xdrv_ws2812.ino
@@ -384,7 +384,7 @@ void ws2812_animate()
   uint8_t fadeValue;

   stripTimerCntr++;
-  if (power == 0) {  // Power Off
+  if ((sysCfg.module == SONOFF_LED) && (power == 0)) {  // Power Off
     sleep = sysCfg.sleep;
     stripTimerCntr = 0;
     tcolor = 0;
@@ -512,14 +512,14 @@ boolean ws2812_command(char *type, uint16_t index, char *dataBuf, uint16_t data_
     if (data_len == 6) {
 //        ws2812_setColor(0, dataBufUc);
       ws2812_setColor(0, dataBuf);
-      power = 1;
+      if (sysCfg.module == SONOFF_LED) power = 1;
     }
     ws2812_getColor(0, svalue, ssvalue);
   }
   else if (!strcmp(type,"DIMMER")) {
     if ((data_len > 0) && (payload >= 0) && (payload <= 100)) {
       sysCfg.ws_dimmer = payload;
-      power = 1;
+      if (sysCfg.module == SONOFF_LED) power = 1;
 #ifdef USE_DOMOTICZ
       mqtt_publishDomoticzPowerState(index);
 #endif  // USE_DOMOTICZ
@@ -578,7 +578,7 @@ boolean ws2812_command(char *type, uint16_t index, char *dataBuf, uint16_t data_
     if ((data_len > 0) && (payload >= 0) && (payload <= 9)) {
       sysCfg.ws_scheme = payload;
       if (sysCfg.ws_scheme == 1) ws2812_resetWakupState();
-      power = 1;
+      if (sysCfg.module == SONOFF_LED) power = 1;
       ws2812_resetStripTimer();
     }
     snprintf_P(svalue, ssvalue, PSTR("{\"Scheme\":%d}"), sysCfg.ws_scheme);

```

davidelang, 2017-04-26:
could you make this into a PR please?

joecotton, 2017-04-26:
Done

arendst, 2017-05-14:
PR rewritten and now supports all modules as an extra "virtual" relay. Will be released in next version (5.1.1).

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2