arendst/Tasmota #387: H801 controls

cobra85ec, 2017-04-25:
how do i control

Staars, 2017-04-25:
Just set the PWM-value via MQTT (0-1023). You can mix RGB that way. I used Homebridge an Node-Red and have a fully working RGB-Strip. It's not that hard.

cobra85ec, 2017-04-25:
can u give me a eg plz...

cobra85ec, 2017-04-25:
OK ...never mind..... i got it

Staars, 2017-04-25:
Well, I can only try to explain my config. 
I use: Homebridge, Homebridge-mqtt, Node-Red ... and of course some iOS-devices at home to control it. 
If you don't use iOS/HomeKit, then I can give only some rough information to you.

cobra85ec, 2017-04-25:
One more question.... what do u use in node-red to pick the color u want.... i got it working with slide.. but qould love to use the color picker

Staars, 2017-04-25:
I get the color from any HomeKit-App.
I goes that way:
Homekit->Homebridge->Homebridge-mqtt->my Node-Red-Flow.
Im my Node-Red-flow:
I subscribe to Homebridge-mqtt and get the HSV-value. Then I convert HSV to RGB and RGB to PWM (that is only multiplying with 4). The PWM-values get published to the H801 via MQTT.

It sounds much more complicated than it is. 
Will you use the color picker from node-red?

cobra85ec, 2017-04-25:
here is my swetup...
RASPBERRY PI3
Mosquitto
NODE_RED
-- I DONT USE HOMEKIT
JUST NODE-RED UI DASHBOARD
i was useing node red to sen pwm with value for each color,,,...
i wasnted to used a simple color picker that would just change the color i pick...

Staars, 2017-04-25:
Okay, I understand.
I will try it out for myself inside Node-Red (without Homekit), but I can’t do it before tomorrow.
When I get it to work, I will report here.

davidelang, 2017-04-25:
If you are asking about the configuration of node-red to have a color picker, 
that's a question you would have to ask the node-red developers.

Tasmota has a minimal GUI, but it's really designed to be managed via mqtt, so 
you can set it via commands (see the commands page for all the details), and 
tasmota is pretty software agnostic as far as things like node-red, homekit, 
openhab, etc go. The wiki has some 'getting started' info, but really, 
configuration of the management software is going to be best dealt with by 
talking to the people who wrote it..

cobra85ec, 2017-04-25:
ok thank u Staars

Staars, 2017-04-26:
Okay, it's even easier with the Node-Red color picker, which gives us already R,G and B.
Do the following:
Take a color-picker and wire it to an function node. Use this code snippet:
`msg1 = {payload: msg.payload.b*4};
msg2 = {payload: msg.payload.g*4};
msg3 = {payload: msg.payload.r*4};

return [msg1,msg2,msg3];`

Use 3 outputs in the function node.

Connect every output to a corresponding mqtt-publish-node, which should publish to "cmnd/YOURH801/PWMx", where x is 3,4 or 5 or whatever PWM-pin you use.

This is a short and hacky approach, but you can extend it as you like.

cobra85ec, 2017-04-26:
trying nothing working... can u export ur flow plz

cobra85ec, 2017-04-26:
i got the mqtt for each color pwm3,4,5 but the funtion dont work

cobra85ec, 2017-04-26:
ok i got it... Thank You very much

davidelang, 2017-04-26:
It sounds as if you have solved this. If not, please re-open the issue.

closed 2017-04-26 by davidelang