arendst/Tasmota #388: I can not use the user_config.h settings

chenfengfeng, 2017-04-26:
user_config.h
```
#define CFG_HOLDER             0x20161209
#define SAVE_DATA              1
#define SAVE_STATE             1
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "192.168.1.199"
#define WIFI_GATEWAY           "192.168.1.1"
#define WIFI_SUBNETMASK        "255.255.255.0"
#define WIFI_DNS               "114.114.114.114"
```
but my ip is 192.168.1.115
---
I can not clear the original data, long press 4 seconds to restore the settings did not work

davidelang, 2017-04-26:
did you restart after doing the 4-second pause?

\

chenfengfeng, 2017-04-26:
@davidelang sorry,i forgot to modify CFG_HOLDER

closed 2017-04-26 by chenfengfeng

davidelang, 2017-04-26:
you should not need to modify CFG_HOLDER if you explicitly tell the device to 
reload the settings (the 4 second button hold, etc)

chenfengfeng, 2017-04-26:
@davidelang can you help me the question #389