arendst/Tasmota #389: can'no use the home-assistant

chenfengfeng, 2017-04-26:
home-assistant configuration.yaml:
```
mqtt:
  broker: domus1
  port: 1883
  client_id: home-assistant-1
  keepalive: 60
  username: pi_mqtt
  password: chenfengfeng
  protocol: 3.1
  birth_message:
    topic: "tele/hass1/LWT"
    payload: "Online"
    qos: 1
    retain: true
  will_message:
    topic: "tele/hass1/LWT"
    payload: "Offline"
    qos: 1
    retain: true

switch:
  - platform: mqtt
    name: "sonoff-1"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true
```

Sonoff-Tasmota user_config.h:
```
#define PROJECT                "sonoff"
#ifdef USE_MQTT_TLS
  #define MQTT_HOST            "m20.cloudmqtt.com" 
  #define MQTT_FINGERPRINT     "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  
  #define MQTT_PORT            20123              
  #define MQTT_USER            "cloudmqttuser"    
  #define MQTT_PASS            "cloudmqttpassword"
#else
  #define MQTT_HOST            "domus1"        
  #define MQTT_PORT            1883            
  #define MQTT_USER            "pi_mqtt"       
  #define MQTT_PASS            "chenfengfeng"    
#endif

#define MQTT_CLIENT_ID         "DVES_%06X"       

#define SUB_PREFIX             "cmnd"            
#define PUB_PREFIX             "stat"            
#define PUB_PREFIX2            "tele"            
                                                 
#define MQTT_TOPIC             PROJECT           
#define MQTT_GRPTOPIC          "sonoffs"         
#define MQTT_BUTTON_RETAIN     0                 
#define MQTT_POWER_RETAIN      1                 
#define MQTT_SWITCH_RETAIN     0                 

#define MQTT_STATUS_OFF        "OFF"             
#define MQTT_STATUS_ON         "ON"              
#define MQTT_CMND_TOGGLE       "TOGGLE"          

// -- MQTT - Telemetry ----------------------------
#define TELE_PERIOD            300               

// -- MQTT - Domoticz -----------------------------
#define USE_DOMOTICZ                             
  #define DOMOTICZ_IN_TOPIC      "domoticz/in"   
  #define DOMOTICZ_OUT_TOPIC     "domoticz/out"  
  #define DOMOTICZ_UPDATE_TIMER  0  
```

davidelang, 2017-04-26:
are you getting sonoff messages to the mqtt server?

chenfengfeng, 2017-04-26:
@davidelang no,i can'no open mqtt,home-assistant messgae:
```
The following components and platforms could not be set up:
switch.mqtt
mqtt
Please check your config
```

davidelang, 2017-04-26:
I can't help you with home assistant problems. I'm trying to find out if your sonoff is sending messages to mqtt properly. I use the mqtt dashboard android app to check this sort of thing

chenfengfeng, 2017-04-26:
thanks! I will continue to look for reasons

closed 2018-05-08 by stale[bot]