arendst/Tasmota #391: Cannot connect while on AC power

Cruisecontrol2016, 2017-04-27:
I know this is gonna sound similar to issue #267 but had a few different symptoms and would appreciate any feedback.

When I connect via serial everything work correctly. When I connect AC power, the green light flashes once and nothing else happens and the pressing the button has no effect. 

When first got my sonoff, I didn't flash is right away and tested it with the eWeLink app and everything worked. Also a few days ago, I was having connectivity issues, but I was still able to have the button work and cycle the relay. 

Again any suggestions on things to try would be greatly appreciated.

davidelang, 2017-04-27:
can you have the serial connected while you plug it in to power and see what the 
logs show?

Cruisecontrol2016, 2017-04-27:
I can try it, any suggestions to do it safely?

davidelang, 2017-04-27:
as long as you are not using a sonoff pow, just keep your fingers away from 
everything and make sure you don't have any bare wires touching anything.

Cruisecontrol2016, 2017-04-27:
I connected to AC power and the logs did not show anything. I cycled the AC power several times.

davidelang, 2017-04-27:
If you don't get anything from the serial port at all, then I would reflash it.

Cruisecontrol2016, 2017-04-27:
alright, I will do that. Also, when I disconnected the serial but left it on AC power, the board stayed properly connected via the webserver. I will update after I reflash.

davidelang, 2017-04-27:
wait, I thought you said that you were saying that it wouldn't connect at all on 
AC power, now you are saying that it remains connected.

please re-state the problem you are having.

Cruisecontrol2016, 2017-04-27:
When I power it over AC, it will not connect and the button doesn't do anything. (That is the problem I am having). I did as you suggested in this order:
1. Connected via serial (board powered on and connected via serial and webserver)
2. Applied AC power (board stayed connected and nothing showed in the logs)
3. Disconnected serial while leave AC power applied (board stayed connected via webserver)

davidelang, 2017-04-27:
when you power it from the serial port, you should see log data on the serial 
port (or if not, you should configure logging to give you logs)

then power it from AC and connect the serial port to the computer after it's 
powered on from the AC and see what you get in the logs
k

Cruisecontrol2016, 2017-04-27:
I didn't see any errors from the logs. I did restart the board and got the logs below. The only thing that seemed strange to me was the time stamps jumping

00:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_3BC225, GroupTopic sonoffs) Version 4.1.3
00:00:00 Wifi: Connecting to AP1 BPfife1 in mode 11N as sonoff-0549...
00:00:04 Wifi: Connected
00:00:04 mDNS: Initialized
00:00:04 HTTP: Webserver active on sonoff-0549.local with IP address 192.168.1.68
00:00:06 MQTT: Attempting connection...
00:00:07 mDNS: Query done with 0 mqtt services found
04:01:54 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds
04:02:00 RSLT: tele/sonoff/UPTIME = {"Time":"2017-04-27T04:02:00", "Uptime":1}
04:02:05 MQTT: Attempting connection...
04:02:06 mDNS: Query done with 0 mqtt services found
04:02:13 MQTT: Connect FAILED to domus1:1883, rc -2. Retry in 10 seconds

Still if I attempt to connect just with AC power, the sonoff will not connect to wifi. As soon as I connect my usb to my RFTI board, the sonoff will connect to wifi.

davidelang, 2017-04-27:
that sounds like there is a problem with the power supply on the board. Try 
measuring the voltage on the 3.3v pin when it's connected just to AC power.
k

Cruisecontrol2016, 2017-04-27:
I get 3.28 V between the 3.3v pin and ground.

davidelang, 2017-04-27:
that should be enough to power the chip. If you power it up on AC and then 
connect the serial port, what happens? (can you connect the serial port except 
for the 3.3v line?)

On Thu, 27 Apr 2017, Cruisecontrol2016 wrote:

> Date: Thu, 27 Apr 2017 06:41:24 -0700
> From: Cruisecontrol2016 <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f1b5884dc2758e41873641588e7dcdb2b92282e0092cf000000011519b70
>     492a169ce0d63dca9@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Cannot connect while on AC power (#391)
> 
> I get 3.28 V between the 3.3v pin and ground.
>
>

Cruisecontrol2016, 2017-04-27:
If I do that, everything appears to work. I can log onto the web server and no error messages in the log from the serial port.

davidelang, 2017-04-27:
even if you don't connect the 3.3v line from your serial cable?
k

Cruisecontrol2016, 2017-04-27:
correct. I did not connect the 3.3v line from my serial cable. But still sometimes when I apply AC power only, the board will not connect to wifi (the green light flashes but nothing else happens), or the board will connect to wifi and I can log onto the web server for a minute or two then the web server stops working and the board does not show up on my ap.

davidelang, 2017-04-27:
but if you connect the tx/rx/ground pins it works properly every time?
k

Cruisecontrol2016, 2017-04-27:
Not everytime. I just try it again and got the following error and I could hear the board click.

:00:00 APP: Project sonoff Sonoff (Topic sonoff, Fallback DVES_3BC225, GroupTopic sonoffs) Version 4.1.3

 ets Jan  8 2013,rst cause:4, boot mode:(3,1)

wdt reset

I have gotten this error 4 times in a row cycling AC power to the board.

chriskmn, 2017-05-01:
Hi, did you measure the logic voltage while plugged in ? I had a similar issue and found out that the voltage sometimes dropped below 2.7. I resoldered the low voltage side of power supply. Then it was ok.

Cruisecontrol2016, 2017-05-03:
Yes. I did check the voltage while plugged in. I got a stead 3.28 V

boko6, 2017-06-10:
Same issue here, the one thas fails on AC is the only one that i have tested with ewlink (like 2 weeks working) app, I tested the sonoff with firmwares 5.1.4 and 5.1.6

goprojojo, 2017-06-10:
Same issue here with Sonoff POW. 
When on AC there is no activity (no blinking leds)
When on serial (3,3v) the sonoff connect to WIFI without problems

davidelang, 2017-06-10:
see the troubleshooting page, this is not an unusual problem.
k

boko6, 2017-06-10:
in my case, without AC power, it Works perfect all the time. with AC power it doesn't connect to wifi but button Works.
On Serial i can reflash it or connect to the web an upgrade it by OTA, LED Works all the time.
It was working properly with itead firmware, this behaviour with AC power is after flashing.

alphagold, 2017-08-11:
Hi, the same for me
----------
in my case, without AC power, it Works perfect all the time. with AC power it doesn't connect to wifi but button Works.
On Serial i can reflash it or connect to the web an upgrade it by OTA, LED Works all the time.
It was working properly with itead firmware, this behaviour with AC power is after flashing.
----------

I noticed that while it is powered by AC, if I take a multimeter and i measure the volt between pins 3.3 and GND, the measurement is correct and as long as I leave the connected leads pin the "sonoff" runs correctly. 

When I unplug them, I lose the wifi connection, reconnect them the connection back to work and all the function also. 

I suppose that, the connection with the multimeter eliminates some disturbance. 

Anyone of you can try this tips, for me worked whit all the firmware that i tryed......

Anyone of you can tell me what I should put between the 3.3v and GND pins to make the effect of the multimeter's leads pin permanent?

Thank you

glsf91, 2017-11-05:
I have the same problem.
But when adding a capacitor it looks like it is solved. You can here see the capacitor just below D5.
[image]
Picture is from a Sonoff unit where capacitor is included now by Itlead.
If you look at the schematic of the [TH10](https://www.itead.cc/wiki/images/3/39/Sonoff_TH10A%2816A%29_schmatic.pdf) you see it is also in the diagram.

I'm uncertain about the right value. Looks like 1 nF 400V if I understand the schematic correctly.
I just used 100nF 50V for testing but have to find the right one.

TD-er, 2017-11-05:
But how could it have been working with the default firmware?

Looks as a pretty good idea to include a capacitor there (I also think it is a Ceramic 400V 1 nF )
However, make sure you do not use too much solder here, since it is between the ground of the low-voltage part and the net powered side of the circuit. When using too much solder, you may bring those two too close to each  other, which may lead to undesired effects when connecting external sensors.

glsf91, 2017-11-05:
I have no idea why it was working with default firmware. I tried not very long with the default firmware.
I tried with esp easy and Sonoff-Tasmota. Problem was the same with both.

alphagold, 2017-11-05:
[image]

alphagold, 2017-11-05:
I just received a new sonoff and comparing it to the old one, I confirm that that condenser is missing

alphagold, 2017-11-23:
Do y know where I can buy a Ceramic 400V 1 nF as in the picture?

glsf91, 2017-11-24:
I bought them in a local electronic part store. They ordered them somewhere in UK.

stronkje, 2017-12-17:
I have exactly the same problem ESPurna, but also Tasmota, software is working correctly when connected to USB, but very instable when connected to 230V mains AC.
My sonoff RF DOES have the capacitor as mentioned above.
I will try to resolder some parts, and hope the best.

w4tsn, 2018-05-07:
Hey there! I have the same problem that programming works and while connected to the programmer the module works as expected. As soon as I disconnect from the programmer and connect it to AC it does nothing. I checked with a multimeter that 230 V is connected to the input. On the programmer connections there is no voltage.

## What I tried already

I already tried to use platform 1.5 instead of 1.7 and also explicitly choosing 1.7 instead of the default.

## My system

I'm running Fedora Linux 27 with Visual Studio Code and PlatformIO extension.

I've cloned the most recent version of the master branch d875dae47ad4c9da7d186459bca2fc7f3b18e9a9.

My settings are as follows:

```
[common] ; ************************************************************
; *** Esp8266 core for Arduino version
platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
;platform = espressif8266@1.7.0  ; v2.4.1
;platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
;platform = espressif8266![photo_2018-04-23_10-25-40](https://user-images.githubusercontent.com/2271467/39672541-5228e81c-512c-11e8-8ad3-5752ca8526f3.jpg)

framework = arduino
board = esp01_1m
board_flash_mode = dout

; *** Fix espressif8266@1.7.0 induced undesired all warnings
build_unflags = -Wall

build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
  -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
;  -DUSE_CONFIG_OVERRIDE

; *** Fix Esp/Arduino core 2.4.x induced Tasmota unused floating point includes
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
upload_speed = 115200
;upload_speed = 512000
upload_resetmethod = ck
upload_port = /dev/ttyUSB0
``` 

The board looks like this:
[image]

Open-Elec-RU, 2018-09-09:
Same problem here.  TASMOTA -flashed without issue.  Works as expected when powered by usb fetid adapter, web interface and all - but doesn't ping when commented to mains.
Worked perfectly with Ewelink

Open-Elec-RU, 2018-09-09:
multimeter show 3.3v between gnd and vcc pins when powered by AC

carloszacu, 2018-10-23:
I have  both styles of board (with and without the capacitor) and I can confirm that the one with the capacitor works perfectly and the one without the capacitor doesn´t work on A/C, can anyone confirm adding the capacitor solves te issue?

Open-Elec-RU, 2018-10-23:
Happened to me also.  One Basic would only "work" with DC applied, button working/relay clicking, the other with the button NOT working, and both NOT working when powered by AC.

My scientific solution was to trash the darn things and install new $5 basics.

glsf91, 2018-10-23:
> I have both styles of board (with and without the capacitor) and I can confirm that the one with the capacitor works perfectly and the one without the capacitor doesn´t work on A/C (works perfectly on flash pins feed), can anyone confirm adding the capacitor solves te issue?

In my case: yes

alphagold, 2018-10-23:
Also in my case

Inviato da smartphone Samsung Galaxy.
-------- Messaggio originale --------Da: glsf91 <notifications@github.com> Data: 23/10/18  19:50  (GMT+01:00) A: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com> Cc: alphagold <alphagold@inwind.it>, Comment <comment@noreply.github.com> Oggetto: Re: [arendst/Sonoff-Tasmota] Cannot connect while on AC power (#391) 

I have both styles of board (with and without the capacitor) and I can confirm that the one with the capacitor works perfectly and the one without the capacitor doesn´t work on A/C (works perfectly on flash pins feed), can anyone confirm adding the capacitor solves te issue?

In my case: yes

—
[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arendst/Sonoff-Tasmota/issues/391#issuecomment-432349917",
"url": "https://github.com/arendst/Sonoff-Tasmota/issues/391#issuecomment-432349917",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [arendst/Sonoff-Tasmota] Cannot connect while on AC power (#391)",
"sections": [
{
"text": "",
"activityTitle": "**glsf91**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@glsf91",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"arendst/Sonoff-Tasmota\",\n\"issueId\": 391,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/arendst/Sonoff-Tasmota/issues/391#issuecomment-432349917"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 220629618\n}"
}
],
"themeColor": "26292E"
}
]

carloszacu, 2018-10-24:
just added the capacitor  and still getting the same issue, i´ll keep trying to see if I can fix this...

carloszacu, 2018-10-25:
any link to where I could buy the capacitor? I live in Colombia so it´ll be nice if it´s from Aliexpress or Amazon

ascillato, 2018-10-25:
Will be cheaper to buy another sonoff than the capacitor + shipping.

Also, you can make a request for a refund to your incomplete device.

carloszacu, 2018-10-25:
since I bought it from a local retailer I don´t know if that'll work

wizbrewery, 2018-11-28:
Hi, Anyone know of a supplier in the UK where the sonoff works with mains input with tasmota firmware?

sheminasalam, 2019-05-25:
> ok I have some news, struggling and making resolder and tests, I fried one of the Sonoffs ( :( ) buuuuuuut since it was already dead, I harvest it´s organs, C15 and C4 specifically (TBH i don´t have much electronics knowledge but I noticed that those capacitors are fisically the same as the one on the older model) and I give it a go and solder it where it should and IT WORKED!!!!!
> 
> so if you have some fried sonoffs around, don´t throw them away, take those capacitors, you´ll need them 👍

Can you explain in detail what solved your issue?

sheminasalam, 2019-05-25:
I had the same issue with my first generation sonoff device tried many methods inculding soldering capacitors. But finally solved it by soldering 3.3v ,RX and Tx together..it solved all the issues. sonoff works, connects to wifi...all same as normal.

daniele-athome, 2019-08-25:
This capacitor seems to be very important. Shouldn't we mention this in the device wiki page?

TD-er, 2025-01-22:
If it is a power related issue, then it might make a difference how quickly the WiFi is turned on after boot.
This can be firmware-dependent, when the RF calibration takes place. The RF calibration depends on the Vcc and if this isn't stable yet, then the calibration may be completely off.

sfromis, 2025-01-22:
Very old Sonoff Basic devices are known to just go bad. I've had two of those die on me. Really not good devices.