arendst/Tasmota #392: Bricking of the device for firmware 5.0 (but not 4.x)

Staars, 2017-04-27:
Hi,

i played around with the code and was able to (kind of) brick my Sonoff SV.

My goal was to implement an ARTNET receiver (heavily based on the work of RSTEPHAN) and it even worked somehow (not in realtime).
Then the device crashed and after reconnecting power it went into an reboot cycle with a clicking relais. 
At that moment I was on firmware 5.0 and reflashing did not help in the first moment. After that I flashed a simple arduino sketch, that worked without problems. Then I flashed TASMOTA 4.2 which was no problem too. After that I wanted to update to 5.0 OTA and I had the same reboot cycle again.

So I suppose it has something to do with the new EEPROM feature. Is it possible to wipe it?

Thank You,
Christian

arendst, 2017-04-27:
What's on the serial interface during the reboot cycle.

Command reset  2 would clear all settings.

Staars, 2017-04-27:
Well, it rebooted very fast. 
I try to reconstruct the error. It can take some moments ....
(I just tried to reset the config via Web-Interface on 4.2)

Staars, 2017-04-27:
Okay, here is the serial output. BTW, reset of the config in 4.x via webinterface did not help.

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Project sonoff Sonoff_SV (Topic sonoff_SV, Fallback DVES_A0047A, GroupTopic sonoffs) Version 5.0.0

Exception (0):
epc1=0x40107279 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont

sp: 3fff1ce0 end: 3fff2010 offset: 01a0

>>>stack>>>
3fff1e80:  40107c3e 40101eac 3fffc200 4000050c
3fff1e90:  4020bb11 00000030 40107c3e 4000050c
3fff1ea0:  00000002 3fff0378 00000000 ffffffff
3fff1eb0:  40107c3e 00000000 00000000 3fff0fe8
3fff1ec0:  3fff0da0 3fff079c 3fff09a0 4021bd80
3fff1ed0:  40202fcb 00000030 00000018 4000050c
3fff1ee0:  401074fa 00000030 00000018 ffffffff
3fff1ef0:  401074e8 0002c6f5 00000000 3fff2044
3fff1f00:  0000000e 00000002 00000000 fffffffe
3fff1f10:  ffffffff 3fffc6fc 00000001 00000000
3fff1f20:  00000000 3fffdad0 3fff0fe8 00000030
3fff1f30:  00000000 3fffdad0 3fff0fe8 00000030
3fff1f40:  ffffffff 3fffc6fc 00000001 3fff0ff0
3fff1f50:  00000000 3fffdad0 3fff0fe8 00000030
3fff1f60:  6369706f 6e6f7320 5f66666f 202c5653
3fff1f70:  6c6c6146 6b636162 45564420 30415f53
3fff1f80:  41373430 7247202c 5470756f 6369706f
3fff1f90:  6e6f7320 7366666f 65562029 4020b9d4
3fff1fa0:  2e35206e 00000000 3fff053c 40204752
3fff1fb0:  feefeffe feefeffe 3fff0ff0 00000000
3fff1fc0:  3fffdad0 3fff0fe8 40202f1c 3fff0ff0
3fff1fd0:  00000000 feefeffe 3fff0fe0 3fff0fe8
3fff1fe0:  3fffdad0 00000000 3fff0da0 4021f429
3fff1ff0:  3fffdad0 00000000 3fff0fe0 40202f68
3fff2000:  feefeffe feefeffe 3fff0ff0 40204130
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

Staars, 2017-04-27:
Okay, I'm on 5.0 again.

Here's what I did:
Re-flashed with 4.1.3 with PlaformIO via USB-UART.
In the web console used: cmnd/sonoff/reset 2 (as You adviced me).
Repaired Wifi (now I learned, that the button-press-commands are very useful).
Re-flashed 5.0 OTA.

Everything seems to work again. Thank You very much.

Christian

nacbooth, 2017-04-27:
Would someone be kind enough to explain "...If you compile your own download the new linker to your IDE or Platformio base folder..."  what this involves?    Relative newbie who's not come across this term (linker) in any of my Arduino/ESP adventures so far.

arendst, 2017-04-27:
See wiki > Prerequisite

nacbooth, 2017-04-27:
Ahh... Thank you.  One place I hadn't looked.  Many thanks for a wonderful tool to control all those gadgets.

closed 2018-05-08 by stale[bot]