arendst/Tasmota #398: Sonoff Touch LED

LexPrima, 2017-04-29:
The Itead website adverts this with "Its LED backlight lets you easily find and turn on the switch at night." However tasmota seems do interchange these states. The LEDs are off when the switch is off.

LED_POWER is active at the moment. Would LED_OFF fix this or does it turn the LED off completely?

lxz81, 2017-05-01:
I have seen a possible solution on [another website](https://www.bountysource.com/issues/43123915-sonoff-touch-how-to-change-the-led-behaviour) by changing GPIO_LED1_INV to GPIO_LED1 in the sonoff/sonoff_template.h file. Haven't tried this though..

khcnz, 2017-05-01:
That website is just a front for github anyway - thats this ticket https://github.com/arendst/Sonoff-Tasmota/issues/214

arendst, 2017-05-01:
The touch has two leds. One connected to the same control line as the relay (in the middle) so that when the relay is on the led is also on. Unable to control it seperately from the relay.

The second led is the wifi indicator which my software controls with both commands ``ledstate`` and ``ledpower``.

Those are all leds available on my touch so other than the wifi led I see no possibility for a "LED backlight".

LexPrima, 2017-05-01:
Thanks for clearing things up

closed 2017-05-01 by LexPrima

TRIROG, 2017-05-24:
Quick question:
@LexPrima make sure to set the  `Module type` to `Sonoff Touch` and not `Sonoff Basic` in `Module configuration` menu. Then the WiFi led is on when the switch is off (to help you find the switch in the dark :)) and vice versa

However i would prefer for the WiFi led to somehow reflect the status of the switch i.e.  to turn on once the swtich has established WiFi or MQTT connection and reported Online to LWT topic, and turn off when offline? Maybe blink while attempting to connect?

madradrakie, 2019-12-13:
@glsf91 can you make a picture or explain what you mean with "connect to GND?". Do you mean hardwire the track to GND? Can you show us how this looks like?

glsf91, 2019-12-13:
@madradrakie 
Yes, I mean hardwire the track to GND.

madradrakie, 2019-12-21:
@glsf91 , can you maybe share a picture how you hardwired the lower part to GND. I would like to know were and how you attached to wire.

glsf91, 2019-12-21:
> @glsf91 , can you maybe share a picture how you hardwired the lower part to GND. I would like to know were and how you attached to wire.

You can see it at the red arrow, the big soldering you see.

madradrakie, 2019-12-22:
@gls91, it worked! Thanks. Do you also know how to tweak the sonoff touch T2? In that version the buttons are at the right and left.

glsf91, 2019-12-22:
@madradrakie, I don't have a T2 so I don't know.

madradrakie, 2019-12-22:
@glsf91 thanks for your help anyway. 4 of my Touch T1 are now modified and glowing in the dark! :-)

I think the T2 works at the same way as the T3, the only problem is that the connection for the left LED is underneath the touch button which i can not reach.