arendst/Tasmota #401: Sonoff S20 inverted State LED

chriskmn, 2017-05-01:
Hi,

I just created my first and second Tasmota Sonoff device and I am impressed and thanlkful to the developer(s) !!!

The Sonoff S20 green LED (state) seems to be controlled inverted by the ESP8266 (pull down). So switched to ON turns it off.
The blue LED (power?) is controlled via a transistor. Here ON is ON.

So maybe anyone can correct/invert the green LED if it makes sense ?!

Cheers, Chris

davidelang, 2017-05-01:
There are two ways of changing how this LED works

one is to change the template in sonoff_templates.h to change LED1 to LED1_inv 
(or change the module type to wemos and define all the gpios in the config 
rather than using a template)

The other is to use the ledstate modes to change what happens with the LED (see 
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands )

On Mon, 1 May 2017, chriskmn wrote:

> Date: Mon, 01 May 2017 04:27:11 -0700
> From: chriskmn <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f71ceb8bab589ef21a6465403d9286d32dfe049a392cf00000001151edd8
>     f92a169ce0d6fa923@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: [arendst/Sonoff-Tasmota] Sonoff S20 inverted State LED (#401)
> 
> Hi,
>
> I just created my first and second Tasmota Sonoff device and I am impressed and thanlkful to the developer(s) !!!
>
> The Sonoff S20 green LED (state) seems to be controlled inverted by the ESP8266 (pull down). So switched to ON turns it off.
> The blue LED (power?) is controlled via a transistor. Here ON is ON.
>
> So maybe anyone can correct/invert the green LED if it makes sense ?!
>
> Cheers, Chris
>
>

arendst, 2017-05-01:
According to the schematic and my observation the blue led is controlled by the same hardware line (PWM0) as the relay so if the relay is ON the blue led is ON too.

The green led, connected to PWM1 can be controlled by the ``ledstate`` and ``ledpower`` commands. In my case using command ``ledstate 1`` the green led follows the relay (as expected).

I see no need to change any inversion...

chriskmn, 2017-05-01:
Thank you. I changed the S20 definition in the "sonoff_template.h" from  GPIO_LED1_INV to  GPIO_LED1.
Now it works fine for me :-)

closed 2017-05-01 by chriskmn