arendst/Tasmota #402: Wemos D1 mini Powerlevel?

timtimsson, 2017-05-01:
Hi,
all my Wemos D1 mini have got  a Powerlevel between Vcc 2,72-2,78. 
Is this correct or a little bit to low?
Thanks

davidelang, 2017-05-01:
that seems quite low to me.

arendst, 2017-05-01:
Mine are low too. It looks to be normal for Wemos D1 mini.

timtimsson, 2017-05-01:
... only for my understanding, 
if i have a level on tele from 2,7V on the sonoffLEDPanel, then breakdown the sonoff device?!)
If i measurement between the wemos pin:
GND-3.3=> 3.02V
GND-5=> 4.47V
where can i find the 2.7V?
Thanks

timtimsson, 2017-05-01:
ups, my ElectroDragon Module shows: 0.224V ?

arendst, 2017-05-01:
Dive into the wonderfull world of esp8266 ADC and enjoy it's dynamic functionality...

On the Sonoff Led it provides a great means of comparing within it's family with functional (> 2.7V) and disfunctional. On the wemos I've never see any problems with ADC voltages of 2.7V

arendst, 2017-05-04:
The ADC input on Sonoff is not connected.

The ADC input of the Wemos is connected with a 100k resistor to Gnd which will lower the ADC value considerably resulting in your findings.

The Electrodragon is worse by connecting the ADC with a 1k resistor to Gnd resulting in even lower values.

closed 2018-05-08 by stale[bot]