arendst/Tasmota #403: Sonoff Dual Problem

jiafengwang, 2017-05-02:
Hello
I have a Sonoff Basic and a Sonoff Dual.When I flashing the Tasmota to Sonoff Dual, aduino says it flashed successfully and then connect to (AC) power, it just blinks the blue led about some 30 times and then a long blue led, then turns off the led, and after a couple of seconds repeates.
But the same firmware is working well with my Sonoff Basic.
Can anybody know about this?
Thanks a lot.

gajotnt, 2017-05-02:
Make sure you have your wifi settings right, had that problem once and i was making a mistake in the SSID of the wifi.

jiafengwang, 2017-05-02:
Yes, I am sue. The ssid and password is correct because the Sonoff Basic is working normally.
Only Sonoff Dual can not work.
Even can not access AP Mode (WIFI Manager) to set wifi configration by 4 short presses.

gajotnt, 2017-05-02:
:-/ i also had some troubles configuring a Dual, try another usb cable and turn off serial loggin.
After some tryes and using another usb cable i got it to finally work

jiafengwang, 2017-05-02:
[image]

gajotnt, 2017-05-02:
It seams it cannot connect to the Wifi, do you have wifi signal were you're installing it?
And again, double check in the user_config if the AP settings are right.

jiafengwang, 2017-05-02:
I am sure the SSID and Password is correct. The router  also works well. 
I even tried to compare these two behavior : one is modifying wifi setting by typing my own ssid and password, the other is keeping default setting.
Both can flash successfully, and show the same logs above.  The ssid both are same named"indebuurt1"  even if I typed my own SSID.

davidelang, 2017-05-02:
if you power the device from your serial cable does it come up and connect to 
the wifi?

you would be far from the first to have a module that has trouble connecting to 
wifi while on mains power.

jiafengwang, 2017-05-02:
Both serial and AC power are showing the same erro.

davidelang, 2017-05-02:
If you can't connect to your wifi, even when powered by the serial port, then 
unless there is a signal strength issue, it's either a config error or there is 
something odd about your wifi.

if you do the 'scan for networks' option, does your network show up and what 
signal strength did it show?

jiafengwang, 2017-05-02:
My Sonoff Basic is working well at the same time with the same router. Only Sonoff Dual can not work. Even can not access AP Mode (WIFI Manager) to set wifi configration by 4 short presses.

jiafengwang, 2017-05-03:
[image]

jiafengwang, 2017-05-03:
I found the version of this Sonoff Dual is "ver 2.0"
Is there something different from the old?

davidelang, 2017-05-03:
the wifi is a feature of the esp8266 chip, other than antenna design, it 
shouldn't be affected by anything else.

jiafengwang, 2017-05-03:
I have tried flash another new Sonoff Dual just now, OMG,  exactly the same problem.
[image]

davidelang, 2017-05-03:
set logging level to 4 (more debugging) and see if there is any more info from the logging when it attempts to connect.

what does the 'scan for networks' show?

jiafengwang, 2017-05-03:
I solved this problem finally by using "esptool.py --port COM4 erase_flash" command to erase the flash.
Thank you all.
WOW~WOW~WOW~

closed 2017-05-03 by jiafengwang

davidelang, 2017-05-03:
If that solved your problem, then you didn't have the config set the way you 
thought you did. See the troubleshooting page about why user_config.h changes 
may not be active.