arendst/Tasmota #405: Reduce time for sensor report

mikenolet, 2017-05-02:
#define TELE_PERIOD            300               // [TelePeriod] Telemetry (0 = disable, 10 - 3600 seconds). 

I have tried to reduce this value to update the sensor reporting time.  Is this the correct paramater to have a temp sensor report temp changes via MQTT?

davidelang, 2017-05-02:
yes, reducing this number will generate the tele/ reports more frequently

mikenolet, 2017-05-02:
This defined only in the user_config.h file?
I have set the value to 30 but still see messages after 5 min after flashing.

`// -- MQTT - Telemetry ----------------------------
#define TELE_PERIOD            30               // [TelePeriod] Telemetry (0 = disable, 10 - 3600 seconds)`

davidelang, 2017-05-02:
no, you can configure it via a command or in the web page (configuration -> 
logging)

look in troubleshooting for why your change to user_config.h didn't get applied.

David Lang

mikenolet, 2017-05-03:
That did the trick. Thanks.

closed 2017-05-05 by arendst

mrohner, 2017-11-08:
wHAT DID THE TRICK?

bachya, 2018-03-23:
@mrohner Check out this page: https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Specifically, entering this command:

```
TelePeriod 60
```

...will change the telemetry period to 60 seconds (from the default 300).

msandik, 2019-10-27:
Can the telemetrie be changed via mqtt command?

ascillato, 2019-10-27:
Yes.

Explained in wiki at commands:

cmnd/yourtasmotatopic/teleperiod 60

Where 60 are seconds

bhengeveld, 2020-05-26:
I gather from the console and the commandslist that the smallest value of Teleperiod is 10 seconds? Is there a way to shorten this? I'd like to switch a dustcollector using something like current or voltage going through a smartplug (or eventually shelly) as a trigger. It would be ideal to have that switch on after 1 or 2 seconds.

ascillato, 2020-05-26:
If you want a mqtt message when a sensor reaches a value or a threshold, you need to use rules. Please, check the docs at rules. There are examples of exactly that. Thanks.

ascillato, 2020-05-26:
If you need further assistant on that, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

bhengeveld, 2020-05-27:
Thank you for pointing me in the right direction. So much to learn! :)