arendst/Tasmota #407: Help needed for active switch on GPIO14

mrohner, 2017-05-03:
Today my lamp is attached to a 220V socket that is switched. As I cannot just add a Sonoff in between (The Sonoff would loose power when I use the Switch), I came up with the following idea:

   regular 220V socket --------------- Sonoff- Lamp
                                         |                                               The switched 220V can the toggle 3.3V
                                         |                                               or 0V on GPIO14
switched-220V socket-Trafo - 3.3V -Sonoff-GPOI14-in

The Switch 09-12 possible on GPIO14, I understand, work when the pulled-up GPIO14 is connected to ground. What I need though is a switch that would have GPIO14 not pulled up so I can toggle the Light on the Sonoff Basic with a latched switch either 3.3V  or 0V that I would input on GPIO14.

gajotnt, 2017-05-03:
Your question is (to me at least) a little confusing, but i have a sonoff basic connected to a two way normal wall switch, using pull up resistors and condensors to prevent signal interferance due to the long cable that goes from the wall swith to the sonoff.
And all works great, one is connected to GPIO 14 and the other to GPIO 3.

So to resume, i soldered 2 10k resistors from 3.3v to the GPIO, and on the wall switch i put a 100uF condensor. On the sonoff configured the GPIO as switches.

mrohner, 2017-05-03:
The wall switch is 220V!

gajotnt, 2017-05-03:
Not if you change the connection :)

mrohner, 2017-05-03:
I do not want to dismantle my electric installation.

gajotnt, 2017-05-03:
[image]

gajotnt, 2017-05-03:
Ahhh you're using diferent way. sorry did not see the photo you loaded

mrohner, 2017-05-03:
Thanks, but this is exactly what I cannot do. My switch has to stay 220V.

stefanbode, 2017-05-07:
Than you are in trouble :-) If your switch OFF/ON the power on the SONOFF this cannot work. The SONOFF must have 100% power on to operate (should be obvious). if you want to use 220V AC instead of the 3.3V DC to switch the lamp you need some electronics inside to translate. Typically you need a transistor, some diodes a small capacitor and a voltage split. Anyhow I'm not providing any further help because I really don't like this configuration.

mrohner, 2017-05-08:
I decided to use a 220V relais now. It switches GPIO14 to GND. A bit an overkill but it works

closed 2017-05-08 by mrohner

nunoronconcouto, 2018-01-04:
Hi 
Anyone can help please...?
I have sonoff with tasmota firmware....last version...i configure GPIO14 to toggle the relay ( Button 2 )..but when i connect GPIO14 with GND nothing happend...
Any ideia why??

davidelang, 2018-01-04:
do you have a relay2 for button2 to affect?

otherwise, configure GPIO to switch1, not button2

On Thu, 4 Jan 2018, nunoronconcouto wrote:

> Date: Thu, 04 Jan 2018 14:33:30 +0000 (UTC)
> From: nunoronconcouto <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f020e7eb8495d814f38f6fdc10efbd7d9a4417b6892cf000000011665fd3
>     a92a169ce0d77498a@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: Subscribed <subscribed@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] Help needed for active switch on GPIO14
>     (#407)
> 
> Hi
> Anyone can help please...?
> I have sonoff with tasmota firmware....last version...i configure GPIO14 to toggle the relay ( Button 2 )..but when i connect GPIO14 with GND nothing happend...
> Any ideia why??
>
>
>
>