arendst/Tasmota #408: Syslog disabled at every hour

terba, 2017-05-03:
Hi!

I was investigating some connection problems, when I saw the following in the remote syslog when grepping for `re-enabled:`

```
May  3 00:11:01 sonoff user ESP-SYSL: Syslog logging re-enabled
May  3 01:11:01 sonoff user ESP-SYSL: Syslog logging re-enabled
May  3 02:11:01 sonoff user ESP-SYSL: Syslog logging re-enabled
May  3 03:11:00 sonoff user ESP-SYSL: Syslog logging re-enabled
May  3 04:11:01 sonoff user ESP-SYSL: Syslog logging re-enabled
May  3 05:11:01 sonoff user ESP-SYSL: Syslog logging re-enabled
... and so on
```

So at every hour at 01 min 00 seconds syslogging gets disabled on the Sonoff because of an undeliverable syslog message. After SYSLOG_TIMER seconds (600) it gets re-enabled. On the syslog host cron hourly is running exactly at this time, but I stopped it, and the result is the same (it doesn't had any hourly jobs anyway). I the meantime MQTT traffic is working in that 10 minutes (can be seen in the mosquitto logs) What can that be?

The environment is:
- Sonoff TH 10
- Webserver disabled
- MQTT with TLS enabled
- Domoticz enabled

Should I connect over serial to monitor this, or do you have any tips?

Thanks!

arendst, 2017-05-03:
If ``portUDP.beginPacket`` in file ``support.ino`` at line 972 fails it disables syslog for ten minutes. So in your case this seems to happen for a (network) reason we do not know yet...

terba, 2017-05-04:
I changed LogHost from domain name to IP address, and the problem disappeared. I don't think, that my DNS server has any problems, it runs without issues for years. So there must be something with the ESP side.

I changed MqttHost also to IP address to be safe.

closed 2018-05-08 by stale[bot]