arendst/Tasmota #410: S20 Reset/Reboot issue

chriskmn, 2017-05-03:
Hello,

after two days using S20 with tasmota I saw  a very high number of resets in the web info and related to that an inceased number of flash write counts. I also heard that the flash chips of ESP are of low quality with little write cycles.
I already added a capacitor to the reset pin and a second electrolytic one to 3.3v.

Is this a "normal" situation or do I have a hardware problem ?
Can I stop flash writing (e.g. PowerOnState=0) ?

Thanks for help,

  Chris

Info:
Program version	5.0.0
Build Date & Time	2017-05-01T13:55:02
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	15 Hours
Flash write count	728
Boot count	335
Reset reason	Software/System restart
Friendly name 1	Sonoff
 
AP2 SSId (RSSI)	malibu (68%)
Hostname	sonoff-1-0797
IP address	192.168.0.15
Gateway	192.168.0.1
Subnet mask	255.255.255.0
DNS server	192.168.0.1
MAC address	5C:CF:7F:9F:43:1D
 
MQTT Host	192.168.0.180
MQTT Port	1883
MQTT Client &
 Fallback Topic	DVES_9F431D
MQTT User	mosquitto
MQTT Topic	sonoff-1
MQTT Group Topic	sonoffs
Emulation	None
mDNS Discovery	Enabled
mDNS Advertise	Webserver
 
ESP Chip id	10437405
Flash Chip id	1327328
Flash size	1024kB
Program flash size	1024kB
Program size	446kB
Free program space	556kB
Free memory	25kB

arendst, 2017-05-03:
See wiki commands. Play with savestate and savedata

chriskmn, 2017-05-08:
Second device online and again after one day more than 1300 resets:

Sonoff Basic Module
Sonoff
Program version	5.0.0
Build Date & Time	2017-04-29T18:37:41
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	1 Hours
Flash write count	1397
Boot count	1366
Reset reason	Exception
Friendly name 1	Sonoff

Savedata, savestate and poweronstate all to 0.

Can the resets be caused by turning off wifi by night ?

ecsfang, 2017-05-08:
Check the **wificonfig** command on the wiki. As I understand, the Sonoff will reboot every minute if no WiFi is found (1366 times corresponds to about 23 hours).
Try **wificonfig 4**, then the sonoff will try to connect again, but not reboot.

chriskmn, 2017-05-08:
Thank you, sounds reasonable. I'll try it out :)

closed 2017-05-09 by chriskmn