arendst/Tasmota #411: Sonoff Pow Yesterday

Satellietje, 2017-05-03:
I have a strange thing with the Sonoff Pow's I have 4 and everyone has the same problem.

I have run on the Pow's 4.1.2. The communication by the Pow to Domoticz is ok, ut Domoticz shows yesterdays usage also today
.
[image]
Image is from the washing machine Pow:
Washing machine Used on sunday
Monday not, should be about 0.017
Tuesday used
Wednesday not, should be about 0.017

If I look at the Pow self, it is correct.
[image]

All other flow meters work well, only the Sonoff Pow does not work ok with today.
I've updated domoticz to the latest version, but it does not fix the problem.

If I use more energy than yesterday, he just adds the difference to today and counts well further.
It seems that the pow told to late to Domoticz that it's a day later or something. The clock is exactly right in Domoticz and in the Pow.

arendst, 2017-05-04:
I'm afraid that your findings are the result of my daily resetting of the energy value. Domoticz probably wants to receive an incrementing value (as provided by my smart meter) and calculates the daily values from it.

That would need some additional code providing a total energy use counter. I sleep on it...

Satellietje, 2017-05-04:
I'll keep waiting until you wake up again :-)

mihalski, 2017-05-05:
Can Domoticz maybe accumulate the usage using the period value?

arendst, 2017-05-05:
I've already implemented a total energy counter which I believe will solve the Domoticz problem and also provides a nice addition to the counter count ;-).

To be released soon.

arendst, 2017-05-05:
@Satellietje what Sensor Type did you select for the Pow Energy usage in Domoticz? ``Usage (Electric)`` or ``Electric (Instant+Counter)``?

mihalski, 2017-05-05:
How long will the counter accumulate before it resets?

arendst, 2017-05-05:
Good question. You caught me just in time. Have to rewrite this counter...

mihalski, 2017-05-05:
I have my moments. My immediate thought was that if it accumulates forever it will overrun or similar.

arendst, 2017-05-05:
Roll-over at about 40MWh so that should take a while...

mihalski, 2017-05-05:
If my calculations are correct, that would be $9,844 of electricity here :P Plus fees and surcharges.

Satellietje, 2017-05-05:
I use (Instant + Counter)

@Satellietje what Sensor Type did you select for the Pow Energy usage in Domoticz? Usage (Electric) or 
Electric (Instant+Counter)?

Satellietje, 2017-05-06:
@arendst, That was a quick update, I installed him on one Pow.
Tomorrow I know if it works.

mlamie, 2017-05-06:
I have installed 5.0.4 and see the below in the Sonoff console. My question is why does it take several minutes to update the power usage?

`22:33:29 MQTT: domoticz/in = {"idx":160,"nvalue":1,"svalue":""}
22:33:29 MQTT: stat/sonoff/RESULT = {"POWER":"ON"}
22:33:29 MQTT: stat/sonoff/POWER = ON
22:34:40 MQTT: domoticz/in = {"idx":162,"nvalue":0,"svalue":"109;12.7"}
22:34:40 MQTT: tele/sonoff/ENERGY = {"Time":"2017-05-06T22:34:40", "Total":0.013, "Yesterday":0.000, "Today":0.013, "Period":7, "Power":109, "Factor":1.00, "Voltage":226, "Current":0.464}
22:36:58 MQTT: domoticz/in = {"idx":160,"nvalue":0,"svalue":""}
22:36:58 MQTT: stat/sonoff/RESULT = {"POWER":"OFF"}
22:36:58 MQTT: stat/sonoff/POWER = OFF
22:39:58 MQTT: domoticz/in = {"idx":162,"nvalue":0,"svalue":"0;16.9"}`

p.s. I have created 2x a dummy
- Switch | IDX 160
-  Usage (Electirc) | IDX 162

mihalski, 2017-05-06:
How would you imagine it to work? Send updated power usage every second? The web interface updates much more frequently.

The default for MQTT is 300 seconds which amount to 5 minutes. This can be changed in the source in user_config.h under:
```
// -- MQTT - Telemetry ----------------------------
#define TELE_PERIOD            300               // [TelePeriod] Telemetry (0 = disable, 10 - 3600 seconds)
```

Satellietje, 2017-05-07:
Yes, it works! :)
[image]

arendst, 2017-05-07:
@mihalski  and @mlamie try focussing on online commands instead of changing user_config.h and recompile. Most parameters can be changed online. In this case use teleperiod.

@Satellietje glad it works. 

I'll close this issue soon.

mihalski, 2017-05-07:
True enough. The only user_config.h change that I would recommend is the wifi network and password.

closed 2017-05-07 by arendst