arendst/Tasmota #412: Odd behaviour when out of WiFi range

CommodoreWhite, 2017-05-03:
Sorry to add to your load but I've noticed an unexpected behaviour when my device running 5.0.1 is out of range of its WiFi access point.

I've got the covers off my electrodragon at present and noticed if the device struggles to connect to a WiFi, it flashes what I suspect is the gpio 0 led for a few seconds every minute then momentarily switches the relay turning off the load for a mere fraction of a second.. It might have something to do with poweronstate=on. Unless this is intended behaviour, I suggest that the relay (and hence the load) should be left alone while the device tried to (re) establish a network connection.

CommodoreWhite, 2017-05-04:
Hmmm, a similar behaviour was reported in #287

CommodoreWhite, 2017-05-04:
... and #278. I'll do the increased debugging level thing to see if that sheds light on (or off!) the problem when I can free up a device. Regards.

arendst, 2017-05-04:
Wiki commands wificonfig 4

CommodoreWhite, 2017-05-05:
I'll try it, thanks

closed 2017-05-05 by CommodoreWhite