arendst/Tasmota #414: Question: Uptime

mrmin24, 2017-05-04:
Hi,

What is the uptime exactly? Is it only based on power on? If so, can anyone tell me what the reason can be for my values generally be not much higher than 20 mins even though the devices are powered on and online for hours? This is on all my devices. POWs, SVs, 1 channels, wemos.

Thx

davidelang, 2017-05-04:
get detailed logs from your devices and look at what happens around the time 
that the uptime starts.

uptime is the time since the software restarted, if you are restarting every 20 
min, something is wrong.

mrmin24, 2017-05-04:
first look shows the following 

00:00:14 MQTT: tele/42/STATE = {"Time":"1970-01-01T00:00:14", "Uptime":0, "Vcc":2.814, "POWER1":"OFF", "POWER2":"OFF", "POWER3":"OFF", "POWER4":"OFF", "Wifi":{"AP":1, "SSID":"Home", "RSSI":22, "APMac":"AC:9E:17:95:AD:10"}}
00:00:20 HTTP: Logging Seriallog 4, Weblog 2, Syslog 0, Host domus1, Port 514, TelePeriod 300
00:02:00 MQTT: tele/42/UPTIME = {"Time":"1970-01-01T00:02:00", "Uptime":1}
00:05:29 MQTT: tele/42/STATE = {"Time":"1970-01-01T00:05:29", "Uptime":1, "Vcc":2.810, "POWER1":"OFF", "POWER2":"OFF", "POWER3":"OFF", "POWER4":"OFF", "Wifi":{"AP":1, "SSID":"Home", "RSSI":30, "APMac":"AC:9E:17:95:AD:10"}}
00:11:16 APP: Serial logging disabled
00:11:31 MQTT: tele/42/STATE = {"Time":"1970-01-01T00:11:31", "Uptime":1, "Vcc":2.811, "POWER1":"OFF", "POWER2":"OFF", "POWER3":"OFF", "POWER4":"OFF", "Wifi":{"AP":1, "SSID":"Home", "RSSI":32, "APMac":"AC:9E:17:95:AD:10"}}
00:17:33 MQTT: tele/42/STATE = {"Time":"1970-01-01T00:17:33", "Uptime":1, "Vcc":2.813, "POWER1":"OFF", "POWER2":"OFF", "POWER3":"OFF", "POWER4":"OFF", "Wifi":{"AP":1, "SSID":"Home", "RSSI":30, "APMac":"AC:9E:17:95:AD:10"}}

Which you can see by the timestamp (wrong for some reason) the uptime is not changing when expected. 
Also the Teleperiod is 300 so the reporting of tele is off by about a minute

Will keep looking at logs

davidelang, 2017-05-04:
try increasing the log level (more debugging)
k

mrmin24, 2017-05-04:
it is on level 4 Logging Seriallog 4. 

Is the uptime measured in hours? Just saw that the uptime changed exactly an hour after the previous change

21:02:00 MQTT: tele/40/UPTIME = {"Time":"2017-05-04T21:02:00", "Uptime":1}
22:02:00 MQTT: tele/40/UPTIME = {"Time":"2017-05-04T22:02:00", "Uptime":2}

ecsfang, 2017-05-04:
Hi! 
This is as designed. To my knowledge the uptime is incremented every hour, at hh:02.30 (I think). I.e. if you restart on an even hour, the uptime will increment after 2 minutes and 30 seconds, and if you restart 5 minutes past the hour, the increment is done 57 minutes later. So don't expect the uptime to be exact, but rather an indication on the number of hours runtime.
/Thomas

arendst, 2017-05-04:
Thomas is  correct. I expect to measure uptime in hours, not minutes ;-)

davidelang, 2017-05-04:
normally uptime is reported down to seconds.
k

mrmin24, 2017-05-04:
great, thanks for the info. I feel better now :-)

Does it wrap at some stage?

Thanks again

ecsfang, 2017-05-05:
Yes, it will wrap, since we don't have an infinite number of bits. But since a signed int is used (i.e. 31 effective bits), it will wrap first after about 244978 years. So I guess that something more drastic will happen before the counter will wrap... 😝 
Cheers!

closed 2017-05-05 by mrmin24