arendst/Tasmota #415: Odd behaviour with PIR (SR501) plugged in (cross-talks with the on/off switches)

mzealey, 2017-05-04:
Hi there, I'm testing with a Sonoff 4CH model. I flashed a build of 5.0.1 firmware and it's working great with a DHT22 sensor. I've just tried adding a SR501 to it which I have previously tested/configured using GPIO on a Pi so I know it is triggering correctly. Depending on which of the Switch[1-4] I choose to mark it as on the /md page, when the PIR detects something it triggers that relay on the unit (ie setting as "09. Switch1" triggers relay 1, with "13. Switch 4" it does the same for relay 4). This is true even when i disable MTQQ so it's happening internally. However from reading the documentation it seems that this should be independent of the existing switches on the unit. My aim was to send a motion detected signal immediately to MTQQ and from there issue any required actions.

sebastianlorenzen, 2017-05-04:
I think you need to configure a specific switch topic. It then just sends a MQTT message if motion is detected.

mzealey, 2017-05-04:
Ah I see "This command will be received by the device itself and the relay will be set to the defined state. Just like receiving the same command from any other MQTT source." (https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic). That's a bit strange behaviour isn't it? But changing SwitchTopic via a command works (can't see anywhere to change this in the gui)

sebastianlorenzen, 2017-05-04:
yep, you need to set it via mqtt cmnd

mzealey, 2017-05-04:
OK so the actual problem is changed but I don't think this is very user-friendly. Can I request that non-internal switches are by default given a different topic or so so they don't trigger the relays? And/or that the topic is easy to set in the UI (perhaps a dropdown of 'trigger relay X', 'Send a MTQQ notification called "external-switch", 'Ignore' or so?). This would probably as a new item when the type is set to switch in the configuration page.

arendst, 2017-05-04:
All settings are available via web console. Internal switches can have a different topic with buttontopic. See wiki commands.

closed 2018-05-08 by stale[bot]