arendst/Tasmota #416: snoof touch fresh setup

marcelveldt, 2017-05-04:
Hi!

First of all. I have lot's of son off devices with your fantastic firmware on it, running really great.
I also had 2 units of the Sonoff touch, also working fantastic but the wife wasn't happy with touch switches so I had put them on marktplaats.

Before shipping them to the buyer I had reset the units into a fresh copy of your firmware.
The guy who bought the units from me can't get them to work and doesn't have a serial flash cable lying around.

The units are blinking fast (3 blinks) every 15 seconds.

We tried all combinations in the wifi button setup to get the devices into config mode but so far no luck.

Before I have no choice but to refund the units, do you have any clue what's going on ?

Is there a specific pattern for the led blinking to know in what state they're in ?

Thanks a lot for your help!

You may answer in dutch btw if you like

arendst, 2017-05-04:
The unit is most likely in search of a wifi network. Without a serial connection the only option you have is selecting the correct amount of button presses to start the internal access point. See wiki button function to find out the correct amount of button presses.

The led blinks are just the reboot blinks.

Btw if the buyer has a router with WPS button let him try that feature too.

marcelveldt, 2017-05-04:
thanks, will try a bit more and else will go there with a serial cable :-)

marcelveldt, 2017-05-07:
All fixed now.... I made the stupid mistake of flashing them as 8266 instead of 8285....

closed 2017-05-07 by marcelveldt