arendst/Tasmota #417: Publish LWT on TOPIC change

mrmin24, 2017-05-05:
Would it be possible to send an Offline message to the old topic when TOPIC is changed, before or just after changing TOPIC?

The reason being that when TOPIC is changed, the old topic is never set to Offline before the new TOPIC comes back online. So it looks like 2 devices are online, but only one is.

arendst, 2017-05-11:
Sending an empty message with previous topic will be implemented in next release (unless there is a majority voting for just sending an offline message...)

mrmin24, 2017-05-11:
Maybe a offline message and then a empty one?

arendst, 2017-05-11:
I make it configurable by the user; either select Offline OR empty message ;-)

mrmin24, 2017-05-11:
Haha, OK. Sorry for trouble.

I think you should add a donate button on your wiki😊

mrmin24, 2017-05-12:
Tested v5.0.7

Thanks

closed 2017-05-12 by mrmin24