arendst/Tasmota #418: Additional switch mode type for those wanting to disable the relay

mihalski, 2017-05-05:
I use my Sonoff Pow units for the sole purpose of monitoring ALL the devices plugged in at a particular location. I VERY much do not want to accidentally turn them off. Would it make sense to have a SwitchMode that disables the switch from being linked to the relay?

Or would changing the SwitchTopic be the better solution to this?

arendst, 2017-05-05:
With the switch you mean the button on the Pow? In that case ButtonTopic would partly solve your problem as a single press can then be detoured. A double press still will switch the Pow though.

What about sending the power command? Should that also be impossible or is it just the button you want to disable?

mihalski, 2017-05-05:
The button for sure, and the toggle switch in the webui.. I once accidentally mis-clicked and brought down an entire room full of equipment. I guess disabling the power command would also be good just in case. If it's easier to implement this as a compile time option then I'd be happy with that.

There's definitely a use for power monitoring with relay to help control power usage, but I think the Sonoff Pow is equally useful for pure power monitoring (without risk of accidentally switching things off).

arendst, 2017-05-05:
Implemented in next release as ``poweronstate 4`` turning relay always on and disable any relay control

mihalski, 2017-05-05:
Thank you so much! That's fantastic.

closed 2017-05-05 by mihalski