arendst/Tasmota #420: PIR with sonoff basic..?

cobra85ec, 2017-05-05:
any one have any info to connect a PIR sensor to a sonof basic..?

davidelang, 2017-05-05:
see the wiki page on connecting additional things to a sonoff

sebastianlorenzen, 2017-05-08:
you can also have a look here. https://groups.google.com/forum/#!topic/sonoffusers/GAv0qnjA04A there are some pictures and discussion

gnkarn, 2017-08-27:
this may have been answered before , but is it is clear that a good example is needed , as there are many people looking  for this information .

if we need to connect a PIR to , gpio14 as input , and configure gpio14 for button2 , then , how we read the status of this new button2 , which topic?, which command to request status , ?
i tried , everithing that could possible make sense , no luck so far ,
i would appreciate if somebody could help , thankyou

jhlweb, 2017-10-26:
It's not very well documented. 
But you have to set a switch1 to switch4 as the state for a GPIO pin.
send the command Switchmode(number of the switch) 1 and set the SwitchTopic to 1 or a topic string.

This what is did:
Connect the PIR on/off pin to a GPIO pin. In my case i use a Wemos D1, i use D2= GPIO4
```
http://<SONOFF-IP>/cm?cmnd=Gpio4 10
http://<SONOFF-IP>/cm?cmnd=SwitchMode2 1
http://<SONOFF-IP>/cm?cmnd=SwitchTopic Sonoff-clock1
http://<SONOFF-IP>/cm?cmnd=Restart 1
```

The output to MQTT is:
```
19:30:42 MQT: cmnd/Sonoff-clock1/POWER2 = ON
19:30:45 MQT: cmnd/Sonoff-clock1/POWER2 = OFF
```

closed 2018-05-07 by stale[bot]