arendst/Tasmota #422: Change TEMP_ADJUST, Press_ADJUST

timtimsson, 2017-05-05:
Hello,
my bme280  is working with a little offset - is it possible to fix this ?
PressAdj ->  %value%+-100
Best regards

arendst, 2017-05-05:
Your next bme280 will have another offset. No guarantee it's lineair either...

davidelang, 2017-05-05:
I think what he's asking for is a per-sensor adjustment config option.

mihalski, 2017-05-05:
I'm more interested in how to calibrate these sensors? How can you know that the OTHER sensor you are using isn't the one that's offset?

timtimsson, 2017-05-05:
offset, a good question... for 
temperature, i use a  analog human clinical thermometer to compare
pressure, humi - i use a official outdoor weatherstation, not far from me, with a dht/esp8266 outside my car  with 12V/USB charger...
Please do not laugh about me.... :)))

davidelang, 2017-05-05:
ice water bath and boiling water bath (the boiling water bath needs to be 
adjusted for your altitude)

calibrate one sensor and then use it to calibrate your others.

mihalski, 2017-05-05:
I suspect that if I throw the sensor in boiling water it may malfunction :/

And ice water bath won't necessarily be 0 degrees Celsius will it?

davidelang, 2017-05-05:
I don't remember which sensor is which, but I'm pretty sure that the waterproof 
ones are rated up above boiling water.

an ice water bath will be 0C, if it was lower, it would be solid ice, if it was 
higher, it would melt the ice. It's only at 0C that the two states can co-exist.

Similarly with a boiling water bath, liquid water at seal level pressure can't 
go above 100C, any higher than that and it becomes vapor, any lower and it's not 
boiling. So heat water to a rolling boil and it will be 100C (at sea level, 
adjust for your alititude)

zissouz, 2017-09-09:
It seems there is a general issue with the BME280 see here for fix

[https://github.com/letscontrolit/ESPEasy/commit/1d1d69a13ff0ecc5422bb3b900107bb3013f389b](url)

Id be really grateful if we could get this in, it seems to be a wider issue.

Sjaakafhaak, 2017-10-21:
I was just searching for the same thing timtimsson would liked to have.

Is there away to adjust temp?
If yes where can i fill in this adjustment?

Thanks in advance

closed 2018-05-07 by stale[bot]