arendst/Tasmota #423: Buttons, switches ang GPIO

brunog69, 2017-05-05:
Hi,
i think that the assignment GPIOs - buttons must be configurable from the web interface.
Now it's fixed but it will be really useful if i could assign a GPIO to a different function.
Example, using SONOFF 4CH:
the standard configuration assign GPIO00, GPIO09, GPIO10, GPIO14 to BUTTON 1, 2, 3, 4
Now i can't, from the config module page, change this.
I think that it would be really useful to assign, for example GPIO09 to a switch or to a sensor.
Doing this i will have 7 GPIOs that i can use!
At the start of the module i will see the standard assignment but, if needed, it's configurable.
What do you think?

Thanks,
Bruno

arendst, 2017-05-05:
There is a good reason I did it this way. Think harder. Hint: Take a look at the source code.

brunog69, 2017-05-05:
Ok, i'll do! Just help me a little bit... where in the source code?

In the meantime another request:
I want to attach a switch that locally will not do anything but through mqtt it's sending its status to openhab. For example a sonoff basic with two switches: with one i can control the relay, the other will toggle a relay on another module.

Thanks for you wonderful work!!!

Bruno

arendst, 2017-05-05:
See file sonoff_template.h and browse the wiki. It's all been done before.

davidelang, 2017-05-05:
sonoff_templates.h

but you can just set buttontopic and the buttons will just send mqtt messages, 
not switch any local relays.
k

brunog69, 2017-05-05:
Thank you!
1) in the wiki: Understanding SwitchMode and SwitchTopic
at the end i read
"Conclusion

SwitchMode=0 controls the relay directly.
SwitchMode=1 sends a message with the default topic to MQTT. This message will get picked up by the device itself and sets the state of the relay accordingly.
While SwitchMode=2 sends a command with a custom topic via MQTT. This will not get picked up unless you configure any device to subscribe to this topic."

If i have understood there is a mistake: It's not switchmode but switchtopic

2) The problem i still have is that switchtopic is global and i can't apply this to just one switch. Right?

Thanks,
Bruno

mihalski, 2017-05-05:
In my experience the MQTT device topic and the SwitchTopic need to be different or you will get weird feedback in the console/logs.

davidelang, 2017-05-05:
yes switchtopic and buttontopic are global.

the page is a wiki, so you can fix the errors you find (you are right, it is 
switchtopic) :-)

davidelang, 2017-05-05:
you are correct, buttontopic, switchtopic, topic, and grouptopic should all be 
different
k

jeylites, 2017-07-24:
I'm new to Tasmota and have inherited an ESP8266-03 module. I would like setup the module as a 3 or a 4 button switch, no relays. Is there any documentation that could guide with the process?

I've read up on [Expanding Sonoff ](https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs) but don't quite understand how the ESP8266 GPIO pins correspond to the script. Because on wiki it lists them as functions.

Where in the script should I make changes?

davidelang, 2017-07-24:
you don't need to change anything in a script. define the GPIOs as switches (in 
the web GUI, or via the commands). That page was written before it was possible 
to switch module types in the config
k

jeylites, 2017-07-24:
Ah, that's awesome. Will give it a shot.

closed 2018-05-07 by stale[bot]