arendst/Tasmota #424: Problem with the new DHT structure and HomeAssistant

FallenLeave91, 2017-05-06:
Hi guys.

So since the DHT22 sensor message includes it´s GPIO i can´t get it to work with HomeAssistant.

So new to the MQTT message is the -14 behind the DHT22.
`{
  "Time": "2017-05-06T19:45:49",
  "DHT22-14": {
    "Temperature": 21.8,
    "Humidity": 55.5
  },
  "TempUnit": "C"
}`

So I just added it into my HA Config

```
- platform: mqtt
  name: "Sonoff Temperatur"
  state_topic: "tele/sonoff1/SENSOR"
  value_template: "{{ value_json.DHT22_14.Temperature }}"
  unit_of_measurement: "°C"
- platform: mqtt
  name: "Sonoff Luftfeuchtigkeit"
  state_topic: "tele/sonoff1/SENSOR"
  value_template: "{{ value_json.DHT22-14.Humidity }}"
  unit_of_measurement: "%"
```

But it just doesn´t work. Any idea?

arendst, 2017-05-06:
Not sure but you need DHT22-14 in your temperature definition like you used for humidity.

FallenLeave91, 2017-05-06:
I tried with - and _ but both do not work.

arendst, 2017-05-06:
Try to find out if the value_template supports - and _ within a JSON value as I don't know.

mihalski, 2017-05-06:
I think this is a case of user blindness to the difference between "-" and "_".

FallenLeave91, 2017-05-06:
@mihalski I just left both "options" in. None of those worked.

mihalski, 2017-05-06:
My Sonoff TH is on 4.0.6 still. Even with updating from the latest 4.x version to 5.x I ALWAYS had to refresh via serial.. It may taker me a while before I can test.. But I'll get back to you. Also I use the AM2032 so I don't know if it will also have the GPIO added to the topic.

FallenLeave91, 2017-05-06:
I´ve downgraded to 5.0.1 which is the last Version withouth the support for more than one DHT22. I´m going to ask the HA guys if the "-" is a problem.

mihalski, 2017-05-06:
You ain't kidding:
```
2017-05-07 09:08:27 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'AM2301' (value: {"Time":"2017-05-07T09:08:27", "AM2301-14":{"Temperature":20.8, "Humidity":60.4}, "TempUnit":"C"}, template: {{ value_json.AM2301-14.Temperature }})
2017-05-07 09:08:27 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'AM2301' (value: {"Time":"2017-05-07T09:08:27", "AM2301-14":{"Temperature":20.8, "Humidity":60.4}, "TempUnit":"C"}, template: {{ value_json.AM2301-14.Humidity }})
```
From these sensors which worked prior to the dash:
```
  - platform: mqtt
    name: "Master Bedroom Temp"
    state_topic: "tele/master_bedroom_th/SENSOR"
    unit_of_measurement: "°C"
    qos: 1
    value_template: '{{ value_json.AM2301-14.Temperature }}'

  - platform: mqtt
    name: "Master Bedroom Humidity"
    state_topic: "tele/master_bedroom_th/SENSOR" 
    unit_of_measurement: "%"
    qos: 1
    value_template: '{{ value_json.AM2301-14.Humidity }}'
```
I'm looking into it now.

FallenLeave91, 2017-05-06:
@mihalski That´s a relief. Thanks for the support!

mihalski, 2017-05-07:
I've found this issue https://github.com/home-assistant/home-assistant/issues/3830.

But the solution provided does not work for me.. Still searching.

FallenLeave91, 2017-05-07:
@mihalski I already had the ". But I saw you added [' '] which works for me.

```
- platform: mqtt
  name: "Sonoff Temperatur"
  state_topic: "tele/sonoff1/SENSOR"
  value_template: "{{ value_json['DHT22-14'].Temperature }}"
  unit_of_measurement: "°C"
- platform: mqtt
  name: "Sonoff Luftfeuchtigkeit"
  state_topic: "tele/sonoff1/SENSOR"
  value_template: "{{ value_json['DHT22-14'].Humidity }}"
  unit_of_measurement: "%"
```

arendst, 2017-05-07:
@mihalski Thnx for updating the wiki. I close this issue now.

closed 2017-05-07 by arendst