arendst/Tasmota #426: Sonoff Basic web interface unavailable.

ian-tait, 2017-05-07:
I have found that the web interface becomes unavailable after about 30 minutes.
All other functions are still available, for example I can send MQTT requests and the relay responds as expected.

Restarting the Sonoff brings all services back including the web interface.
I can do this by removing the power or sending cmnd/sonoff/Restart:99 from Node-Red.

I have monitored the serial interface while logging is on "more detail" but nothing is displayed.
This issue can be replicated over and over again but I am not sure what screen shots or logs I can send you that would give you something to work with.

davidelang, 2017-05-08:
When you say it becomes unavailable, what happens when you try to connect to it?

are you still getting logs at the time this happens? or has the serial logging 
stopped before that point?

ian-tait, 2017-05-08:
Hi davidelang
The web page won't load. 
I can still ping the sonoff and it responds. 
On very few occasions the web site has stopped before the serial logging has stopped but still nothing comes up that would indicate there was an issue until you try and navigate the website. 

Since logging this issue, I have rolled back to 5.0.2 and the issue has not happened again. 
It has been running all night without any issue. 

Hope all this helps?

arendst, 2017-05-08:
Did you try different browsers?

ian-tait, 2017-05-08:
Hi arendst,
This was the first thing I tried.
Safari, Chrome, FireFox & IE.

arendst, 2017-05-11:
When this happens do you have any emulation active (Wemo or Hue)?

ian-tait, 2017-05-12:
I have Wemo running.
However you made a change based on #427 and created 5.0.6, and this doesn't have the same issue.
Now running on this new version and don't have this issue.

This can be closed.

closed 2017-05-18 by arendst