arendst/Tasmota #427: JSON parse issue with SENSOR topic

ian-tait, 2017-05-07:
I am very new to Tasmota but I am having some problems making this work with Node Red.
Since you included the GPIO in the JSON topic #404, I cannot get Node Red to read the payload.

Although the idea of having the GPIO included is great idea, Node Red doesn't like "-" is there any chance it can be swapped out for "_"?

Or can you advise on how to handle "-" in Node Red.

Thank you

stefkoch09, 2017-05-08:
Have the same issue with intergrating in Home Assistant. JSON does not take the "-".
Thanks

ian-tait, 2017-05-08:
Hi stefkoch09

I have see a solutions for home assistant under #424. 
You need to place the sensor in square brackets. 
['DHT22-14']

Hope this helps you out ?

mihalski, 2017-05-08:
and drop the "." in front of said sensor.

stefkoch09, 2017-05-09:
That worked ... Thanks a lot!

mihalski, 2017-05-10:
The OP was referring to Node-RED whereas the solution provided only works for Home Assistant. Might it be worthwhile trading the hyphen for an underscore? Not sure if that would solve the problem.. Or even having it configurable through the interface somehow as a "separator" character.

arendst, 2017-05-10:
As I hate the underscore the hyphen stays for now but I reverted the message in case of a single sensor detected. So if one AM2301 is connected the message will be:
```
{"Time":"2017-05-10T14:01:33", "AM2301":{"Temperature":21.8, "Humidity":39.4}, "TempUnit":"C"}
```
And if more sensors are connected the message will stay as:
```
{"Time":"2017-05-10T13:58:50", "AM2301-04":{"Temperature":21.3, "Humidity":43.6}, "AM2301-14":{"Temperature":21.3, "Humidity":43.6}, "TempUnit":"C"}
```

ian-tait, 2017-05-12:
Thank you, this can be closed

closed 2017-05-18 by arendst