arendst/Tasmota #429: If the MQTT server is offline

Satellietje, 2017-05-08:
I have a Wemos D1 in my car.
There are 4 relay modules connected and 4 physical buttons.
If the temperature is below 4 degrees, the wemos turn on the webasto (heater) in my car. 
But if I'm on my way and I want to turn on the webasto with the physical button I have to press a few times before the Wemos responds. This is because the MQTT server is not found on the way.

How can I bypass the wemos waiting and giving a timeout if the MQTT server is not online?

[image]

davidelang, 2017-05-08:
This is a known bug with the libraries, see the troubleshooting page.

unfortunantly, there isn't really any work-around.

arendst, 2017-05-09:
Or disable mqtt when on the road.

Satellietje, 2017-05-09:
But then I have to log in to the wemos every time i use the heater to turn off the MQTT.
Then it's faster I press the button a few times :-)

khcnz, 2017-05-09:
I'd suggest
a) Change the retry time - its 10 seconds but in your case 5mins would probably be fine
b) Hook a GPIO into when the car is running and disable MQTT when running 
c) I presume when on the road there is no wifi? Could we add a check just before MQTT connect to check wifi is connected? WiFi.status() != WL_CONNECTED?

arendst, 2017-05-09:
Option C is already implemented.

Considering the amount of inconvenience people have with the retry timeout I'll introduce a command to change the 10 seconds to a higher value and leave the consequences to the user.

khcnz, 2017-05-09:
Yes you are correct (of course :)

It does raise an interesting question though, if wifi is not connected why does try connect and stall? Could be related to this arduino issue where still reports connected but RSSI =0  is it is disconnected? https://github.com/arduino-libraries/WiFi101/issues/86

arendst, 2017-05-09:
I don't think the arduino-esp8266 project is using wifi101.

Satellietje, 2017-05-10:
Perfectly resolved in v5.0.6
Thank you 👍

closed 2017-05-10 by Satellietje