arendst/Tasmota #430: Battery for Itead RF remote

WayneManion, 2017-05-08:
I know this is a dumb question. I have a Sonoff RF and the accompanying Itead four-button remote. I flashed it with Tasmota. I follow the instructions:

double-click the button on the Sonoff body (ok)
wait for a quick red flash (ok)
press and hold one of the remote buttons (ok)
wait for three red flashes (never happens)

The remote didn't come with any documentation. The board inside had no markings about what size battery to use. What is the correct battery size and voltage for the standard Itead 433 MHz RF remote?

Thanks

arendst, 2017-05-09:
And here is the dumb answer ;;--))

Open the case of the remote and try to find the battery. Considering it's size it shouldn't be hard to find.

This is a positive note. No means to offend you ;-)

WayneManion, 2017-05-09:
It didn't come with a battery. I just grabbed one that looked like it fit, and I'm not sure if it is the right one.

I'm trying to check and see if wrong battery is the reasons for the non working status.

arendst, 2017-05-09:
Does you remote look like this:

[image]

If not then you may have a remote that doesn't work with the Sonoff RF. I used KlikAanKlikUit remotes that didn't work either.

WayneManion, 2017-05-09:
Yes, it is identical to that one.

arendst, 2017-05-09:
On this remote between button A and B is a small hole. If you press a button the led below the hole should light up. If not then your battery is dead indeed.

WayneManion, 2017-05-09:
The question is: what is the correct battery size and voltage?

arendst, 2017-05-09:
See my latest update

WayneManion, 2017-05-09:
Nevermind, I just saw the second paragraph of your response.

I'm using a brand new 12v battery, all I wonder what the problem is.

Thanks

mihalski, 2017-05-10:
I've ordered 2 of these (I had ordered the Sonoff RF's earlier and totally forgot that they didn't come with a remote). Is it even possible that another remote could work? Frankly I'd prefer a nice clean button rather than a garage remote with 4 buttons :/

milen100, 2017-09-04:
My remote looks exactly as on the picture, but the space for battery inside is too small for 27A battery type.
The battery should be both shorter ( 3-5mm) and 2-4mm less in diameter. Anybody knows what might be this battery type and voltage?

closed 2018-05-07 by stale[bot]