arendst/Tasmota #431: DNS lookup for NTP server every 3-6 seconds

mirko, 2017-05-09:
My Sonoff wall-switch apparently really wants to make sure it runs on time.
Or at least (and to be more precise) it tries to resolve the configured NTP server hostname every 3-6 seconds.
As I don't see any reason for doing so I consider this being a bug (and file it herewith as such :)).

Log from the DNS server:

> # tail -f /var/log/messages | grep -E 'query\[A\] .*.pool.ntp.org from 10.5.5.150'
> May  9 02:01:01 ns.owrt dnsmasq[7600]: 374 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:04 ns.owrt dnsmasq[7600]: 375 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:10 ns.owrt dnsmasq[7600]: 377 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:13 ns.owrt dnsmasq[7600]: 378 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:19 ns.owrt dnsmasq[7600]: 381 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:22 ns.owrt dnsmasq[7600]: 390 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:28 ns.owrt dnsmasq[7600]: 397 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:32 ns.owrt dnsmasq[7600]: 399 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:38 ns.owrt dnsmasq[7600]: 402 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:41 ns.owrt dnsmasq[7600]: 405 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:47 ns.owrt dnsmasq[7600]: 407 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:50 ns.owrt dnsmasq[7600]: 408 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:56 ns.owrt dnsmasq[7600]: 413 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
> May  9 02:01:59 ns.owrt dnsmasq[7600]: 414 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
> May  9 02:02:05 ns.owrt dnsmasq[7600]: 418 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150

davidelang, 2017-05-09:
Is it able to resolve those names? or does the DNS lookup fail?

David Lang

arendst, 2017-05-09:
The default ntp servers are pool.ntp.org, nl.pool.ntp.org and 0.nl.pool.ntp.org. If only the last two NTP servers give these DNS errors just change them to known NTP servers in your region with commands ``ntpserver2`` and ``ntpserver3``. Or delete a server with command ``ntpserver3 0``. See wiki > commands.

mirko, 2017-05-09:
All those DNS queries get proper answers

mirko, 2017-05-09:
My apologies, my regexp cut out the requests for 'pool.ntp.org' - it actually requests for all ntp server entries:

```
# tail -f /var/log/messages | grep -E 'query\[A\] .*pool.ntp.org from 10.5.5.150'
May  9 11:51:37 ns.owrt dnsmasq[7600]: 23448 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:51:40 ns.owrt dnsmasq[7600]: 23449 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:51:43 ns.owrt dnsmasq[7600]: 23452 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
May  9 11:51:46 ns.owrt dnsmasq[7600]: 23453 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:51:49 ns.owrt dnsmasq[7600]: 23454 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:51:52 ns.owrt dnsmasq[7600]: 23455 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
May  9 11:51:55 ns.owrt dnsmasq[7600]: 23456 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:51:58 ns.owrt dnsmasq[7600]: 23457 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:52:01 ns.owrt dnsmasq[7600]: 23458 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
May  9 11:52:04 ns.owrt dnsmasq[7600]: 23462 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:52:07 ns.owrt dnsmasq[7600]: 23463 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:52:10 ns.owrt dnsmasq[7600]: 23464 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
May  9 11:52:13 ns.owrt dnsmasq[7600]: 23471 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:52:16 ns.owrt dnsmasq[7600]: 23472 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:52:19 ns.owrt dnsmasq[7600]: 23473 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
May  9 11:52:22 ns.owrt dnsmasq[7600]: 23476 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:52:25 ns.owrt dnsmasq[7600]: 23483 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:52:28 ns.owrt dnsmasq[7600]: 23484 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
May  9 11:52:31 ns.owrt dnsmasq[7600]: 23485 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:52:34 ns.owrt dnsmasq[7600]: 23486 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:52:37 ns.owrt dnsmasq[7600]: 23487 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
May  9 11:52:40 ns.owrt dnsmasq[7600]: 23488 10.5.5.150/4096 query[A] pool.ntp.org from 10.5.5.150
May  9 11:52:43 ns.owrt dnsmasq[7600]: 23491 10.5.5.150/4096 query[A] nl.pool.ntp.org from 10.5.5.150
May  9 11:52:46 ns.owrt dnsmasq[7600]: 23492 10.5.5.150/4096 query[A] 0.nl.pool.ntp.org from 10.5.5.150
```

arendst, 2017-05-09:
Strange. Just looked at my dnsmasq log and there is hardly an entry for ntp. When there is an entry it looks like this:
```
May  9 10:43:26 sidnas2 dnsmasq[4859]: query[A] pool.ntp.org from 192.168.2.200
May  9 10:43:26 sidnas2 dnsmasq[4859]: forwarded pool.ntp.org to 195.121.1.34
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 149.210.130.166
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 87.195.109.106
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 91.148.192.49
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 213.251.53.123
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 213.109.127.195
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 213.239.154.12
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 95.211.224.12
May  9 10:43:26 sidnas2 dnsmasq[4859]: reply pool.ntp.org is 94.228.220.14
```

joecotton, 2017-05-09:
> All those DNS queries are answered with proper A-records

What are the results of those A-records?

closed 2018-05-08 by stale[bot]

Timmiej93, 2019-05-20:
Has this issue ever been resolved? I have 4 Sonoffs connected in my house, 2 S20's, 2 basics. The S20's are fine, but the basics are showing very similar behavior to what mirko described 2 years ago. Tailing my Pi-hole log:
```
May 20 19:45:30 dnsmasq[11313]: query[A] nl.pool.ntp.org from 192.168.2.3
May 20 19:45:30 dnsmasq[11313]: cached nl.pool.ntp.org is 80.127.119.186
May 20 19:45:30 dnsmasq[11313]: cached nl.pool.ntp.org is 51.15.44.1
May 20 19:45:30 dnsmasq[11313]: cached nl.pool.ntp.org is 37.97.195.195
May 20 19:45:30 dnsmasq[11313]: cached nl.pool.ntp.org is 83.98.155.30
May 20 19:45:30 dnsmasq[11313]: query[A] 0.nl.pool.ntp.org from 192.168.2.4
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 129.250.35.250
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 188.42.216.204
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 109.236.82.30
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 5.79.108.34
May 20 19:45:30 dnsmasq[11313]: query[A] 0.nl.pool.ntp.org from 192.168.2.4
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 5.79.108.34
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 129.250.35.250
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 188.42.216.204
May 20 19:45:30 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 109.236.82.30
May 20 19:45:33 dnsmasq[11313]: query[A] 0.nl.pool.ntp.org from 192.168.2.3
May 20 19:45:33 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 109.236.82.30
May 20 19:45:33 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 5.79.108.34
May 20 19:45:33 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 129.250.35.250
May 20 19:45:33 dnsmasq[11313]: cached 0.nl.pool.ntp.org is 188.42.216.204
May 20 19:45:34 dnsmasq[11313]: query[A] pool.ntp.org from 192.168.2.4
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 185.51.192.34
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 213.109.127.82
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 185.255.55.20
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 213.154.236.182
May 20 19:45:34 dnsmasq[11313]: query[A] pool.ntp.org from 192.168.2.4
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 213.154.236.182
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 185.51.192.34
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 213.109.127.82
May 20 19:45:34 dnsmasq[11313]: cached pool.ntp.org is 185.255.55.20
```

In the Pi-hole GUI:
[image]
The two bottom layers (red and blue) are the two Sonoff basics. The dip you can see at 16:30-ish is when I called `NtpServer# 1` for all three servers on both of the devices. This fixed it temporarily, as you can see.
[image]
As you can see, the NTP servers are called upon a lot. The next domain in the list is google.com, at roughly 600.

If I can provide any more information to help resolve this weirdness, please let me know.

m4dm4xi, 2019-05-29:
I have the exakt same problem but only since I updated to 6.5.0 the day before yesterday. I only noticed due to the same reason as Timmiej93... I have a Pi-hole and suddenly the amount of dns requests has significantly increased.

musgos, 2019-09-17:
Same here. Every minute 3 ntp requests by device

Bookingyo, 2020-01-11:
I am having the exact same issue. 4000+ ntp requests in the past 2 hours for all 3 of my Sonoff s31(Tasmota flashed) devices

arendst, 2020-01-12:
What happens if you change the default NTP servers to servers in your country. I bet it solves this issue.

Or simply install a sntp service on your PiHole and service your devices from there.

Many solutions and many fixes possible.

Timmiej93, 2020-01-12:
Well, I can't speak for the other guys, but I'm Dutch, so unfortunately, you've lost your bet. In my system, I've just excluded the Tasmota devices from my pihole, there's no reason for them to get filtered anyway, so they might was well just use a different DNS service.

newhinton, 2020-05-21:
I have the same problem. I tried time1.google.com and 0.de.pool.ntp.org, both get spammed shortly. I upgraded to 8.3.1.1, still trying to get the time.

If i issue the time command, it get's the wrong time. (basically uptime since 1970).
But even if i set it to roughly the proper epoch, it still tries to resolve the time. 

Interestingly enough, all tasmota devices start behaving this way at roughly the same time.

I remember the last time i had an ntp-problem with pihole, i reinstalled pihole and the problem seemed to be fixed. However, since it restarted just now, i suspect that there is more to it, and that something does not play nice here. Is there any way to increase logging so that i can track the ntp-requests directly on a device?