arendst/Tasmota #434: Web interface relative path URL's instead of absolute URL's

BrettSheleski, 2017-05-10:
I have setup a nginx reverse proxy on my Linux box to forward requests to my Sonoff device (which is controlling my basement lights).  I do this mainly to have a single "portal" page on my Linux box and can do a single ssh port tunneling when not on my home network and have access to everything and not have to remember IP addresses and/or ports of all my various things.

nginx configuration:

```
        location /basement-lights/ {
            proxy_pass http://192.168.0.151/; #IP of my sonoff
        }
```

It seems everything is configured to have absolute paths in the URL's in the html.  This works fine when going directly to http://192.168.0.151/ to access the Sonoff web interface, however when  using the reverse proxy by going to http://192.168.0.2/basement-lights/ (IP of my Linux box) (or http://localhost:9090/basement-lights/ when not at home network and tunneling my Linux box's port 80 to port 9090 of my local pc) none of the buttons work as they're all going to absolute path URL's.

Also, all buttons appear to be in html form elements with method=post with a single button element to submit the form.

Example.

Change the following:

```
<form action="/cn" method="post"><button>Configuration</button></form>
```

With:

```
<a href="cn">Configuration</a>
```

Using the a-tags (or by switching the forms to method="get") would also get rid of the warnings web browsers pop up when simply attempting to refresh a page on the web interface and when navigating forward/backward.

Is there a reason for using forms and buttons instead of simple a-tags?  And is there a reason why not using relative-path URL's?

arendst, 2017-05-10:
Please provide a PR with changes to webserver.ino you like to see. 

Currently I'm not convinced that changing ``<button> </button>`` to ``<a href ... </a>`` will provide a visible button like:

[image]

BrettSheleski, 2017-05-10:
It wont look the same without some simple CSS changes.  I will see if I can get some time to test this out and make a pull request.

closed 2018-05-08 by stale[bot]