arendst/Tasmota #435: POW and node red

rene9900, 2017-05-10:
Hi 
i would like my sonoff POW to post the current and voltage usage etc to node red, so i can make a overview page in my current setup. but i can not figure how to get the data in readable format?
i do get data on /tele/POW/# but im missing some thing because i can not add Voltage and add it to a gauge in node red..

10/5/2017 19.18.21node: bde445a.69c5cb8
tele/POW/ENERGY : msg.payload : string[150]
"{"Time":"1970-01-01T04:04:00", "Total":0.111, "Yesterday":0.000, "Today":0.111, "Period":1, "Power":18, "Factor":0.53, "Voltage":237, "Current":0.148}"

ian-tait, 2017-05-10:
I had a similar issue with reading Temp and Humidity the other day. However the process would be the same. This how I did it but its not to say its the right or best way in doing it.
Someone may have a better way and if they do I would like to know.

In node red I assume you have the input MQTT listening on "/tele/POW/#" this will need to be changed to "tele/POW/ENERGY".
The join JSON along with a FUNCTION.
Inside FUNCTION copy paste this:

`var msg1 = {};
var msg2 = {};

msg1.payload = msg.payload.Voltage;
msg1.topic = 'Voltage';
msg2.payload = msg.payload.Current;
msg2.topic = 'Current';

return [msg1, msg2];`

Make sure you select 2 outputs at the bottom of the function box.

Now add your gauge.
The first output on the FUNCTION will be "msg1" and the second will be "msg2".
You can now add a CHART to Both output if you want that way you can graph the line over a period of time.

Hope this helps.

rene9900, 2017-05-16:
cant get it to work, i get zero reading to my graph...
could you please try copy your nodes so that i can see how it is done?

closed 2017-05-17 by rene9900