arendst/Tasmota #436: Chaning Topic using MQTT keeps device online in old topic

Daddy1991, 2017-05-10:
Hi,

I am programming an control application in UWP for Windows 10 including voice commands support using Cortana. It is starting to get in shape but I am having one issue.

I am using MQTT (broker=cloudmqtt.com). When I restart a Sonoff it is immidiately shown as offline, but when I rename the topic using command: "cmnd/sonoff_standing/topic: sonoff_standing_aap" it starts listening to sonoff_standing_app but it never takes offline the old topic.

Any idea what I am doing wrong or is this a bug?

Kind regards,
Sander

davidelang, 2017-05-10:
what do you mean by "never takes the old topic offline"?

once you configure the sonoff for the new topic, it no longer uses the prior 
topic and no longer sends or responds to any messages with the old topic (unless 
the old topic gets set as the grouptopic)

khcnz, 2017-05-11:
I think this relates to the retained flag on the online/offline messages. When you change the topic I don't think it sends an offline message on the old topic. I guess ideally it would send an offline message and tell the broker to stop retaining the message so it falls off the broker.

mrmin24, 2017-05-11:
See ticket #417

mrmin24, 2017-05-11:
Also it should not just send an empty message to broker in my opinion, because if the controller is not online at that stage for some reason, it will never get the offline message before the message is removed. Maybe the controller should clear the broker and Sonoff just send an offline as per ticket #417?

arendst, 2017-05-11:
Sending an empty message with previous topic will be implemented in next release (unless there is a majority voting for just sending an offline message...)

Daddy1991, 2017-05-11:
In my opinion the device should be listed as offline, unsubscribed, or even removed if this is possible by MQTT standards? I am also not sure if it is a bug in the Sonoff firmware or a bug of the broker if the device does not respond for a long time and should list the device as offline after a timeout?

Maybe a solution would be changing the topic after restart so the device will unsubscribe first.

mrmin24, 2017-05-11:
I believe the broker pings on client ID and not topic. The same client ID is still connected and that is why it is not taken offline. 

This is also not a subscribe issue, it is a publish of LWT problem. If it is possible to select offline message at topic change as arendst suggests that should be ok and if broker restarts in non persistent mode it should remove the LWT

Daddy1991, 2017-05-11:
I've contacted the broker to see if this is the problem and asked to propose a solution.

closed 2018-05-08 by stale[bot]