arendst/Tasmota #437: Sonoff and flash upgrade 1M->4M

TRIROG, 2017-05-11:
What changes should one make to uploading procedure settings described in wiki if one upgraded the flash on Sonoff (1M->4M)?

Should one change the flash size from 1M(64k SPIFFS)  to 4M(1M SPIFFS) or just leave it as it is?

arendst, 2017-05-11:
Settings can be migrated by using webpage Backup Configuration and Restore Configuration.

Flash size change may be any as long as it is 1M(No SPIFFS) or higher. 1M(64k SPIFFS) is not supported anymore.

TRIROG, 2017-05-12:
I understand this is only in case where one is preforming an upgrade and wish to retain settings.
So what should one  set Flash size to if  uploading tasmota for the first time to a new Sonoff with 4M flash?

arendst, 2017-05-12:
I use 1M(No SPIFFS) on all my sonoffs (Including 4M Pow) and 4M Wemos as that's the goal of my software; stay as small as half 1M to be able to OTA upgrade Sonoffs. This also allows for one firmware image for all sonoffs.

Changing to 4M(any SPIFFS) has no advantage for my firmware using the default options. If you enable additional features like TLS (using more code space) it might be usefull to select 4M as then the OTA space is bigger.

Conclusion: Use 1M(No SPIFFS) as much as possible or use 4M(1M SPIFFS) if available and needed.

closed 2018-05-08 by stale[bot]