arendst/Tasmota #438: Can I support the reed sensor?

ChienTing, 2017-05-11:
Can I support the reed sensor?
In the door mounted on the reed sensor
Open the door relay on
Close the door relay off
Thanks

smadds, 2017-05-12:
It's just a switch, @ChienTing, so wire it up like a button. MQTT will fire an update to any subscribers when the state changes, so you can use Node-Red or a home automation package to work out the logic

off14, 2017-05-15:
if i wire it to GPIO14, how to tell sonoff that it is not for relay control?

davidelang, 2017-05-15:
define it as a switch and set switchtopic
k

off14, 2017-05-15:
tks!
can it be done via user_config.h, or only after in console?

davidelang, 2017-05-15:
I think only via commands (console, mqtt, etc)

off14, 2017-05-15:
great - after playing with SwitchMode and SwitchTopic, sonoff now works as two independent devices:

1. switch (pulse switch in my case), can be toggled via wifi or via internal button
2. door sensor - when change status sends ON/OFF to a specified topic. works when gpio14 connected/disconneted to GND via reed sensor

TRIROG, 2017-05-28:
@off14 
Would you mind sharing your configuration - what commands did you use?
I'd like to achieve the exact same thing.

off14, 2017-05-28:
1. you have to to to WEB configuration/configure module and configure GPIO14 as **switch1**
2. go to console on web
2a. type: switchmode 2
2b. type: switchtopic alarm
3. DONE.

now, relay is INTERDEPENDENT from the gpio14.
you can still toggle relay using MQTT (or web). and if door is opened(or closed) you will receive "cmnd/alarm/POWER" message (ON or OFF).
you can toggle the relay using original topic e.g. "/cmnd/sonoff/POWER" but reedswitch becomes independent.

jstorgaard, 2017-08-29:
@off14 What Sonoff module and reed switch did you use for this setup?

sabs01, 2017-09-21:
thanks off14 -  this has been very useful - any idea how to set up GPIO3 or GPIO1 to accept a second and third reed switch like this???

bandric1, 2017-11-02:
CAn someone explain this more detailed i have done it all changed everything but it still doesnt work.

I have a wemos d1 mini runing latest tasmota fw. I can publish to it via openhab/mqtt:
and the relay shield trigers. But sensor reed magnet when closed it publishes to mqtt i can see it in mqtt.fx it publishes ON/OFF state. But no matter what i do or write in items in openhab the change is not visible the button stays the sam so as the icon i have been experimenting for the last week and getting frustrated so much i cannot explain.
This is my read out sensor item. And it doesnt work.

Contact N_D_GarageDoor1  "Garage Door 1 [MAP(en.map):%s]" <garagedoor> { mqtt="<[broker:cmnd/alarm/POWER1:state:default]" }

closed 2018-05-07 by stale[bot]

sirs2k, 2018-05-29:
Hey guys, I have made all the changes as @off14 described but still can't get the state to update automatically in HA.

Reed is independent of the switch and it does update state if moved manually.
But I want to able to update state based on time interval or state update mqtt msg.

I've tried all these three sensors in HA without any luck

```
  - platform: mqtt
    name: "Alarm"
    state_topic: "tele/alarm/SENSOR"
    value_template: "{{ value_json['Switch1'] }}"
    unit_of_measurement: "ON/OFF"
  - platform: mqtt
    name: "Alarm2"
    state_topic: "cmnd/alarm/POWER1"
    value_template: "{{ value_json['Switch1'] }}"
    unit_of_measurement: "ON/OFF"
  - platform: mqtt
    name: "Alarm3"
    state_topic: "tele/garage/SENSOR"
    value_template: "{{ value_json['Switch1'] }}"
    unit_of_measurement: "ON/OFF"
```

Francisk0, 2018-08-30:
Hello, 
I try to make an independent sensor and I follow the steps but I do not get results.
1 you have to to to WEB configuration/configure module and configure GPIO14 as switch1
   go to console on web
2a. type: switchmode 2
2b. type: switchtopic alarm
DONE.

But not work, my sensor can activate relay and don´t have information on console about my sensor.

RuneKyndal, 2018-09-06:
Im trying to do a similar thing.  But on the sonoff seperately.

My garage door only have an open/close button.   I want to know and control the state. Up or down.  Not just "toggle"

Im adding a reed sw to be closed when door is closed

Relay and sensor on gpios.  14,RX

Want to do some logic in rules, switch and backlog 
But it seams lacking in logic options.

(Human readable code)
Mqtt Cmd OPEN.   "If sw1 closed"; Then POWER2 ON; delay 5; POWER2 OFF;
else do nothing
Mqtt Cmd CLOSE. "If sw1 open"; Then POWER2 ON; delay 5; POWER2 OFF;
else do nothing
With sw1 also being sent via tele. For visualizing the state.

Suggestions on how to do the logic part,  independent on the sonoff?

Francisk0, 2018-09-13:
I´m solve my problem with mqtt, now work fine with my pir sensor (independient of Relay)
In Tasmota configuration: GPIO14 -->12 Switch4
In tasmota console type:
switchmode4 1
switchtopic alarma

RuneKyndal, 2018-09-17:
Id like to come up with a solution, independant of the mqtt server though

ffontanac, 2019-04-21:
An old post, anyway very close to what I'm looking for.

I have it correctly working (switchmode, switchtopic... etc), except in case the sonoff reboots and the sensor is triggered, it doesn't boot. I think it reacts as if the button1 would be long-pressed during the boot.

Did someone experienced the same?

This my Status 0
`12:17:25 CMD: Status 0
12:17:25 MQT: stat/Tasmota_1/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota_1"],"Topic":"Tasmota_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"sensor","SwitchMode":[0,1,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
12:17:25 MQT: stat/Tasmota_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:03:23","StartupUTC":"2019-04-21T10:14:02","Sleep":50,"CfgHolder":4617,"BootCount":88,"SaveCount":420,"SaveAddress":"FA000"}}
12:17:25 MQT: stat/Tasmota_1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-03-19T12:47:29","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
12:17:25 MQT: stat/Tasmota_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FFF_2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008069","280500000100000000000000000000000000","00000000"]}}
12:17:26 MQT: stat/Tasmota_1/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
12:17:26 MQT: stat/Tasmota_1/STATUS5 = {"StatusNET":{"Hostname":"Tasmota_1-0509","IPAddress":"192.168.1.51","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"80.58.61.250","Mac":"DC:4F:22:E5:01:FD","Webserver":2,"WifiConfig":4}}
12:17:26 MQT: stat/Tasmota_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E501FD","MqttUser":"paco","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:17:26 MQT: stat/Tasmota_1/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 21 10:17:26 2019","Local":"Sun Apr 21 12:17:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:48","Sunset":"20:50"}}
12:17:26 MQT: stat/Tasmota_1/STATUS10 = {"StatusSNS":{"Time":"2019-04-21T12:17:26","Switch2":"ON"}}
12:17:26 MQT: stat/Tasmota_1/STATUS11 = {"StatusSTS":{"Time":"2019-04-21T12:17:26","Uptime":"0T00:03:24","Vcc":3.423,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:04"}}}

`

ffontanac, 2019-04-21:
> An old post, anyway very close to what I'm looking for.
>
> I have it correctly working (switchmode, switchtopic... etc), except in case the sonoff reboots and the sensor is triggered, it doesn't boot. [...]
Solved!, 

I'm using a R2 V1 sonoff (the new one) that doesn´t have visible GPIO14. Insted of GPIO14 I was using GPIO2 that cannot be low at boot (it doesn't boot when triggered).

Using GPIO03 (Rx) insted is fine.

Don't know why they remove the GPIO14 connector  from the PCB!

n00bster-za, 2022-05-04:
has been a while since someone posted here. I need assistance on getting the GPIO14 to read more than just on or off. I have a gate motor with a dedicated Gate status output. The gate status can be open,close,opening,closing,collision. How do I configure the GPIO to read all those different states and add it to HA.

ascillato2, 2022-05-04:
Hi, this issue is very old and your question is not directly related to this. Please, open a new discussion posting all the information you have (hardware type, tasmota version, what have you tried, etc), or ask in the support chat. Thanks.