arendst/Tasmota #439: some of commands not working

hamwong, 2017-05-11:


arendst, 2017-05-11:
Your ``ButtonRetain`` doesn't work because you disabled mqtt.
All ``SetOption<x>`` will be released in next version.

Regarding migrating from < 5.0 to 5.0 see wiki > Upgrade - Migration Path step 4

hamwong, 2017-05-11:
Thanks for response @arendst, I didn't disable mqtt as it's the only reason why I use tasmota but I do disabled **USE_DOMOTICZ**, will it be the reason? as I only use openhab, **DOMOTICZ** feature is useless for me.

I checked the wiki, I did stored config, issue is not happened right after upgrade, it upgraded successfully  to 5.0.x and info also shows corrected version, issue happened after power cut, after power restore, ESP fail to start and serial shows  lot of characters turns to unicode "square", wifi connection keep fail.

I don't know is this a unique case, I will try on other ESP tomorrow see will it happen again.

hamwong, 2017-05-12:
I don't know is this a bug or not,
I found if I use stored setting, MQTT was disabled in info, (config menu dont have MQTT option)
after I do reset setting, MQTT back to default setting in rom, config menu have MQTT again
I don't know what's going on

closed 2018-05-08 by stale[bot]