arendst/Tasmota #440: GPIO 0 Relay does not work

stefanbode, 2017-05-11:
Really strange, but I also tested the sonoff.ino.bin you attached to 5.0.6. As long as I bind a relay to GPIO 1+ it is working. Only the GPIO0 does not work. Because this is on the Wemos the quite common pin D3 i was wondering why on hell my relay did not work again. maybe you can have a look and comment on it. thx.

arendst, 2017-05-11:
See #377

stefanbode, 2017-05-11:
Hi Arend, I agree on the switch issue, but is it the same for the Relay? The strange this is, I'm pretty sure it was working before. will upload an earlier version to verify.

stefanbode, 2017-05-11:
I just did additional test with wemos mini and nodemcu and some older devices with 5.0.3 firmware. It looks for me that relay on GIO 0 was working in 5.0.3 and is broken in 5.0.4. Neither the nodemcu nor the wemos will work with the "actual" firmware on GIO 0

arendst, 2017-05-12:
Did a test with firmware 5.0.7 using a Wemos D1 mini (clone) and configured a relay on GPIO00:

[image]

Connected the relay like:

[image]

And all works as expected although the relay toggles after power on with ``PowerOnState 0`` due to the external pull up resistor R10 on my version of the Wemos D1 mini:

[image]

What's the exact schematic(s) you are using compared to your real hardware?

stefanbode, 2017-05-12:
My mistake.... sorry to bother...

closed 2017-05-12 by stefanbode