arendst/Tasmota #441: Can not ping the Sonoff

Satellietje, 2017-05-12:
Once in a few weeks I have a Sonoff that is no longer accessible via the web browser, it's not a specific Sonoff but random, once a POW the other time a basic. 
I also can not ping that Sonoff anymore, but MQTT still works fine. I get the Sonoff data and can also switch via mqtt and read the POW data. If i do nothing, the ping and web interface remains unreachable.
If I reboot the access point where this Sonoff is connected to, then the sonoff works fine again.

Is this a known problem?

sebastianlorenzen, 2017-05-12:
Have the same from time to time. I restart my sonoffs every night.

x0010, 2017-05-23:
I'm not sure if my scenario is the same but I upgraded a TH16 yesterday from 4.x to latest 5.1.3.
I was using the `sonoff.ino.bin` directly and it all seemed to work fine. But I constantly lost connection, neither ping, mqtt, or http worked. I had to push on the button to make it work again..for like 10-15 minutes max.

My solution (which I can't know if it was the reason) was to reflash it with `sonoff-minimal.ino.bin` first and then with `sonoff.ino.bin` again. Now it's 100% stable again.

TommySharpNZ, 2017-12-08:
How do you restart them every night?

closed 2018-05-07 by stale[bot]