arendst/Tasmota #442: Usermode

Satellietje, 2017-05-12:
How can I log in with an http command as admin if the switch is set as a user?

http://sonoff/cm?User=admin&password=joker&cmnd=WebServer%202

Does not work?

arendst, 2017-05-12:
You can't!

If webserver is in user mode you cannot perform Admin mode tasks. That's a security measure. The main purpose is MQTT MQTT and MQTT.

Satellietje, 2017-05-12:
Mmm okay. Then I have to flash a sonoff again :-)

closed 2017-05-12 by Satellietje

HobbytronicsPK, 2018-07-09:
There is no need to re-flash the Sonoff. Reset the settings according to our Guide **[Configure Sonoff Tasmota Firmware](https://hobbytronics.pk/config-sonoff-tasmota/)**. Just navigate to **[Reset Tasmota to Default Settings](https://hobbytronics.pk/config-sonoff-tasmota/#Reset-Tasmota-Default-Settings)** section.