arendst/Tasmota #443: Not flashing to Sonoff basic or S20

eggfat, 2017-05-12:
Hi there,
Ive been trying to flash my Sonoff's and Im having no luck.
Any help would be fantastic as Im stuck.
Martyn.

The board Im using
https://www.amazon.co.uk/gp/product/B01K1RQE38/ref=oh_aui_detailpage_o04_s02?ie=UTF8&psc=1

Here is the error.

Checking program size
text       data     bss     dec     hex filename
468177     8468   34200  510845   7cb7d .pioenvs/sonoff-touch-4ch/firmware.elf
Building .pioenvs/sonoff-touch-4ch/firmware.bin
Looking for upload port...
Auto-detected: /dev/cu.usbserial-A600CIVX
Uploading .pioenvs/sonoff-touch-4ch/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
========================== [ERROR] Took 15.27 seconds ==========================
 
================================== [SUMMARY] ==================================
Environment sonoff              [SUCCESS]
Environment sonoff-touch-4ch    [ERROR]
========================== [ERROR] Took 80.99 seconds ==========================

sebastianlorenzen, 2017-05-12:
dit it ever work?
If yes, try to fully wipe the device using this:

Wipe Sonoff on Windows
1. Download the esptool.py program from https://github.com/espressif/esptool
2. Put device in firmware upload mode and clear flash with command esptool.py --port COM3 erase_flash
Before:
Install Phyton:
https://www.python.org/
Install Psearial
https://github.com/pyserial/pyserial
pyserial-master>setup.py install

Jantokil, 2017-05-12:
Same problem here. Last week I had no problems flashing another Sonoff. And now I also get the 255 error in Atom. Is there a way to wipe the device on a mac? I thought it wasn't possible at all. But I'm fairly new to this..

eggfat, 2017-05-13:
Hi sebastianlorenzen,
Thanks for the reply Im also on a Mac.

Any ideas how to wipe the Sonoff on a mac?

Im so close any help would be much appreciated.

Martyn.

sebastianlorenzen, 2017-05-13:
Sorry, no clue. Maybe you find something under the first link. Maybe Google could help.

samuel235, 2017-05-20:
For what it is worth, I've been having issues with my sonoff POWs. They're entering flash mode fine but i can't upload things to it. I get the following in the logging with the Arduino IDE:

`
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM4
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 467712 bytes from C:\Users\Samuel\AppData\Local\Temp\arduino_build_672329/sonoff.ino.bin to flash at 0x00000000
	erasing flash
	size: 072300 address: 000000
	first_sector_index: 0
	total_sector_count: 115
	head_sector_count: 16
	adjusted_sector_count: 99
	erase_size: 063000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 15000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 17% ]
................................................................................ [ 35% ]
................................................................................ [ 52% ]
................................................................................ [ 70% ]
................................................................................ [ 87% ]
.........................................................                        [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
`

I was able to get the flash wipe as suggested by @sebastianlorenzen above. Still no joy for my issue though. Just wanted to let you guys know that the flash wiping is stable right now with no issues with installing on a windows machine.

caedn, 2017-05-26:
I had the same issue when using Atom/PlatformIO. Switching to Arduino fixed it for me. I did not switch TX/RX.

closed 2018-05-08 by stale[bot]